jneilliii/OctoPrint-BedLevelVisualizer #351: [BUG]: Octoprint UI gone blank after BedLevelVisualizer installation

marciuliani, 2021-01-03:
**Describe the bug**

**Expected behavior**

**Debug Log**

**Screenshots**

**Firmware and Version**

jneilliii, 2021-01-03:
Please provide logs from octoprint, I've used this plugin, there are several others that use this plugin, and no-one else is having this problem.

marciuliani, 2021-01-04:
I hope this is what you're looking for

jneilliii, 2021-01-04:
Sorry, attachments don't come through via email. You need to drag the file into a comment here.

marciuliani, 2021-01-04:
I couldn't copy or drag the file over but here is the log itself.  The issue began right after I installed the bedlevelvisualizer plugin on 1/2

2021-01-01 04:04:20,958 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-01 04:04:20,964 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2021-01-01 04:04:20,965 - octoprint.server - INFO - OctoPrint 1.5.2
2021-01-01 04:04:20,972 - octoprint.plugin.core - INFO - 28 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  AstroPrint (1.6.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_astroprint
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  BLTouch Plugin (0.3.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_BLTouch
|  Change Filament Plugin (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_Change_Filament
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  CuraEngine Legacy (1.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_curalegacy
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/errortracking
|  Filament Sensor (1.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament
|  Filament Sensor Reloaded (1.3.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentreload
|  File Check (2020.8.7) (bundled) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_file_check
|  Firmware Check (2020.9.23) (bundled) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmware_check
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/gcodeviewer
|  HeaterTimeout (0.0.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_HeaterTimeout
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  OctoPrint Anywhere (1.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_anywhere
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  SMS Notifier (with Twilio) (0.3.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_smsnotifier
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  UI Customizer (0.1.1.7) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_uicustomizer
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
|  WebcamStreamer Plugin (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_webcamstreamer
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2021-01-01 04:04:20,997 - octoprint.environment - INFO - Detected environment is Python 2.7.16 under Linux (linux2). Details:
|  hardware:
|      cores: 4
|      freq: 1400.0
|      ram: 917016576
|  os:
|      bits: 32
|      id: linux
|      platform: linux2
|  plugins:
|      pi_support:
|          model: Raspberry Pi 3 Model B Plus Rev 1.3
|          octopi_version: 0.17.0
|          throttle_state: '0x0'
|  python:
|      pip: 19.3.1
|      version: 2.7.16
|      virtualenv: /home/pi/oprint
2021-01-01 04:04:20,997 - octoprint.server - INFO - ------------------------------------------------------------------------------
2021-01-01 04:04:25,146 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 691224}
2021-01-01 04:04:35,415 - backoff - INFO - Backing off __get_dev_settings__(...) for 132.1s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-01-01 04:06:47,749 - backoff - INFO - Backing off __get_dev_settings__(...) for 73.8s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-01-01 04:10:12,923 - backoff - INFO - Backing off __get_dev_settings__(...) for 282.2s (ConnectionError: HTTPConnectionPool(host='newstream.getanywhere.io', port=80): Max retries exceeded with url: /api/dev_settings (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x650df3d0>: Failed to establish a new connection: [Errno 110] Connection timed out',)))
2021-01-01 04:14:55,389 - backoff - INFO - Backing off __get_dev_settings__(...) for 281.9s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-01-01 04:19:20,973 - octoprint.server.heartbeat - INFO - Server heartbeat <3
[... cut ...]
2021-01-01 14:04:25,932 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 727225}
2021-01-01 14:05:44,989 - octoprint.util.comm - INFO - Telling the printer to set the busy interval to our "communicationBusy" timeout - 1s = 2s
2021-01-01 14:08:29,244 - backoff - INFO - Backing off __get_dev_settings__(...) for 213.8s (ConnectionError: HTTPConnectionPool(host='newstream.getanywhere.io', port=80): Max retries exceeded with url: /api/dev_settings (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x64c304f0>: Failed t

jneilliii, 2021-01-04:
I see lots of these errors, and that has nothing to do with my plugin. I don't know what that is.

```
2021-01-01 11:05:14,715 - backoff - INFO - Backing off __get_dev_settings__(...) for 162.4s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
```

marciuliani, 2021-01-05:
Odd

This happened after I installed the bedlevelvisualizer, but it must have been a coincidence.  I don't know what the error is below, or what that URL is, but I was able to get my UI back configuring it through the UI customizer.

Sorry about that - I thought there was a correlation between the plugin and my UI disappearing.

Marc

closed 2021-01-05 by jneilliii