jneilliii/OctoPrint-BedLevelVisualizer #352: [BUG]: mesh data with points with more than single digits doesn't collect data.

jneilliii, 2021-01-04:
**Describe the bug**
Using the Snapmaker the mesh report returns points larger that 9 (ie, 10+) and the regex doesn't extract the points properly.

**Expected behavior**
To red the mesh as valid numbers and graph the data.

See this discussion on the forum.

https://community.octoprint.org/t/bedvisualizer-and-snapmaler-2-0/28407?u=jneilliii

jneilliii, 2021-01-04:
I think this is fixed in 1.0.1rc1 release.

deputyd0ng, 2021-01-28:
Ok so I'm no longer timing out. And don't get errors. But there's no mesh visible with the RC release. Just the graph.

jneilliii, 2021-01-28:
Enable the option to use relative z offset and that will bring the mesh down into view. Or change the z graph limits to auto,auto.

PR #398 (2021-02-28): 1.0.1
**Added**
* optional timeout override to @BEDLEVELVISUALIZER command, resolves #314
* settings button on tab for quick access, resolves #349

**Updated**
* README.md adding note that auto bed leveling must be possible on printer, resolves #348

**Fixed**
* allow for probe points that contain more than single digits, resolves #352, resolves #269
* resolve issues with circular beds
closed 2021-02-28 by jneilliii