jneilliii/OctoPrint-BedLevelVisualizer #353: Add plot point markers to the grid so that users can identify unleveled areas easier

Osanchez, 2021-01-04:
Enhancement request

Osanchez, 2021-01-04:
depending on how many plots are made, I think it will be helpful to see where on the grid each plot point is.

jneilliii, 2021-01-04:
When you hover the mouse over the areas on the graph you should see the approximated point on the bed.

[image]

Osanchez, 2021-01-05:
My confusion with this is the x and y axis may be inverted. Would 0,0 be
the front left of the bed always ?

jneilliii, 2021-01-05:
Yes and no. 0,0 could be middle of bed as well for example in delta printers or if that option is enabled in the settings. The plugin does not account for inverted axes because from my experience thus far this doesn't actually exist. If you are able to provide examples in your situation where this truly is the case and you provide a mesh report that shows such and graphs/pictures demonstrating an inverted or rotated axis I'd be glad to try and figure out how to get the graph to display properly relative to that situation.

antonio-gob, 2021-01-14:
some days ago i was able to click on the graph then the nozzle move to that part but now is not possible? was something change?

antonioreyna, 2021-01-14:
some days ago i was able to clic on a part in the graph and then the nozzzle moves to that part, but now is not possible did the plug in changed that?

jneilliii, 2021-01-14:
Nothing should have changed relative to that. You have to enable the move option though first. It's a button in the button bar that pops up. If it's not working please open new issue.

closed 2021-02-08 by github-actions[bot]