jneilliii/OctoPrint-BedLevelVisualizer #354: [BUG]: Not probing

ColePete23, 2021-01-06:
**Describe the bug**
whenever I click "create a mesh" the probe doesn't move around and get a mesh level of the bed. 

**Expected behavior**
The probe should go around and get a feel of where the bed is 

**Screenshots**
Do yo need any?

**Firmware and Version**
Marlin 2.0 *I think*

jneilliii, 2021-01-06:
What are you using as your gcode to send in the plugin's settings? Examples can be found [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md).

ColePete23, 2021-01-06:
[image]
exactly this

ColePete23, 2021-01-06:
hello, you there?

jneilliii, 2021-01-06:
Enable debug logging in the plugin settings, restart octoprint and try again. Then share your plugin_bedlevelvisualizer_debug.log file.

ColePete23, 2021-01-06:
> Enable debug logging in the plugin settings, restart octoprint and try again. Then share your plugin_bedlevelvisualizer_debug.log file.

how to I enable that?

ColePete23, 2021-01-06:
[image]
this is what my settings look like

jneilliii, 2021-01-06:
Yes, it appears you have have debug logging enabled now. Press the button and then go into OctoPrint Settings, click Logging on the left, download the file plugin_bedlevelvisualizer_debug.log, and then drag it into a comment here to upload and share with me.

ColePete23, 2021-01-06:
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/5777308/plugin_bedlevelvisualizer_debug.log)
like that @jneilliii?

ColePete23, 2021-01-06:
well now I feel stupid. All I had to do was home the bed first. once I did that, it worked. 
[image]
but now I have another question. I want the bed to be in the green right?

jneilliii, 2021-01-06:
So you can add the G28 command to your gcode in the plugin's settings and that will take care of that. 

To answer your question, yes, you want to get the variance between the points to be as minimum as possible. It is not always possible to get a perfectly flat bed, which is why Auto Bed Leveling exists, but manually leveling your bed prior to probing is the ideal to make sure that your firmware doesn't have to work so hard to adjust height while printing. Make sure you are probing with your bed heated to printing temperature as well, as that will effect the bed's shape as it expands.

closed 2021-01-06 by jneilliii