jneilliii/OctoPrint-BedLevelVisualizer #355: [FR]: Add G28 to the Marlin gcode example

lightmaster, 2021-01-11:
In the example gcode for Marlin at https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/wiki/Firmware-Specific-GCODE-Examples#marlin-119-with-auto_report_temperatures-enabled-and-ubl, I'd suggest adding `g28` after the `m155` code so the first thing it does before checking the bed level is making sure Home is properly set. I'd also suggest extending the `M155` timer since you're gonna be putting it back to 3 seconds after the bed leveling anyway.

New example gcode would be:
```
M155 S30		; set temperature reporting delay, use a value longer than the time it takes for your leveling command to complete.
G28                     ; Find Home
@BEDLEVELVISUALIZER	; instruct plugin to start recording responses from printer.
G29 T			; report the bed leveling mesh points.
M155 S3			; set the temperature reporting delay back to a shorter time span.
```

jneilliii, 2021-01-11:
I need to kill the wiki examples, as I've actually got this documented as you mention already in.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md

jneilliii, 2021-01-11:
Removed Wiki page and linked back to wiki pages in repo.

closed 2021-01-11 by jneilliii