jneilliii/OctoPrint-BedLevelVisualizer #357: [BUG]: Probing failed in the middel of the probing process

lopio1234, 2021-01-12:
**Describe the bug**
Everytime i start the "Update Mesh" the probing getting started but in the middel of the probing process the probing stops with the notification "Probing failed, close connection"

In the mesh image is the mesh graphic only displayed in half
[image]

**Debug Log**
[2021-01-12 10:34:30,264] DEBUG: mesh collection started
[2021-01-12 13:28:37,376] DEBUG: mesh collection started
[2021-01-12 13:29:02,907] DEBUG: mesh collection started
[2021-01-12 13:37:04,446] DEBUG: Canceling mesh collection per user request
[2021-01-12 13:37:04,447] DEBUG: Mesh data collected prior to cancel:
[2021-01-12 13:37:04,447] DEBUG: []
[2021-01-12 13:37:04,447] DEBUG: Mesh data after clearing:
[2021-01-12 13:37:04,448] DEBUG: []
[2021-01-12 13:39:30,785] DEBUG: mesh collection started
[2021-01-12 13:39:58,363] DEBUG: mesh collection started
[2021-01-12 13:53:28,380] DEBUG: mesh collection started
[2021-01-12 13:59:40,026] DEBUG: ['+2.075', '+1.985', '+1.870', '+1.750', '+1.647']
[2021-01-12 13:59:40,028] DEBUG: ['+4.727', '+4.617', '+4.367', '+4.265', '+4.165']
[2021-01-12 13:59:40,030] DEBUG: ['+0.865', '+0.768', '+0.665', '+0.895', '+0.798']
[2021-01-12 13:59:40,031] DEBUG: ['+1.545', '+1.442', '+1.330', '+1.232', '+0.952']
[2021-01-12 13:59:40,032] DEBUG: ['+1.795', '+2.008', '+1.885', '+1.762', '+1.720']
[2021-01-12 13:59:40,038] DEBUG: {'type': 'rectangular', 'x_min': 0, 'x_max': 220.0, 'y_min': 0, 'y_max': 220.0, 'z_min': 0, 'z_max': 250.0}
[2021-01-12 13:59:40,038] DEBUG: stopping mesh collection
[2021-01-12 13:59:40,039] DEBUG: [['+2.075', '+1.985', '+1.870', '+1.750', '+1.647'], ['+4.727', '+4.617', '+4.367', '+4.265', '+4.165'], ['+0.865', '+0.768', '+0.665', '+0.895', '+0.798'], ['+1.545', '+1.442', '+1.330', '+1.232', '+0.952'], ['+1.795', '+2.008', '+1.885', '+1.762', '+1.720']]
[2021-01-12 14:11:14,968] DEBUG: mesh collection started
[2021-01-12 14:11:42,551] DEBUG: mesh collection started
[2021-01-12 14:14:25,337] DEBUG: ['+1.115', '+1.015', '+1.062', '+0.947', '+0.840']
[2021-01-12 14:14:25,342] DEBUG: ['+1.520', '+1.420', '+1.160', '+1.058', '+0.962']
[2021-01-12 14:14:25,345] DEBUG: ['+1.508', '+1.413', '+1.310', '+1.222', '+1.127']
[2021-01-12 14:14:25,346] DEBUG: ['+1.875', '+1.775', '+1.655', '+1.560', '+1.282']
[2021-01-12 14:14:25,348] DEBUG: ['+1.955', '+1.853', '+1.717', '+1.600', '+1.525']
[2021-01-12 14:14:25,351] DEBUG: {'type': 'rectangular', 'x_min': 0, 'x_max': 220.0, 'y_min': 0, 'y_max': 220.0, 'z_min': 0, 'z_max': 250.0}
[2021-01-12 14:14:25,353] DEBUG: stopping mesh collection
[2021-01-12 14:14:25,356] DEBUG: [['+1.115', '+1.015', '+1.062', '+0.947', '+0.840'], ['+1.520', '+1.420', '+1.160', '+1.058', '+0.962'], ['+1.508', '+1.413', '+1.310', '+1.222', '+1.127'], ['+1.875', '+1.775', '+1.655', '+1.560', '+1.282'], ['+1.955', '+1.853', '+1.717', '+1.600', '+1.525']]
[2021-01-12 14:16:31,547] DEBUG: mesh collection started
[2021-01-12 14:16:59,132] DEBUG: mesh collection started
[2021-01-12 14:19:40,168] DEBUG: ['+0.312', '+0.217', '+0.102', '-0.015', '-0.125']
[2021-01-12 14:19:40,172] DEBUG: ['+2.955', '+2.855', '+0.200', '+0.102', '-0.160']
[2021-01-12 14:19:40,174] DEBUG: ['+2.943', '+2.852', '+2.753', '+2.815', '+2.730']
[2021-01-12 14:19:40,176] DEBUG: ['+3.790', '+3.367', '+3.090', '+2.840', '+2.715']
[2021-01-12 14:19:40,179] DEBUG: ['+3.870', '+3.770', '+3.955', '+4.000', '+3.857']
[2021-01-12 14:19:40,189] DEBUG: {'type': 'rectangular', 'x_min': 0, 'x_max': 220.0, 'y_min': 0, 'y_max': 220.0, 'z_min': 0, 'z_max': 250.0}
[2021-01-12 14:19:40,190] DEBUG: stopping mesh collection
[2021-01-12 14:19:40,193] DEBUG: [['+0.312', '+0.217', '+0.102', '-0.015', '-0.125'], ['+2.955', '+2.855', '+0.200', '+0.102', '-0.160'], ['+2.943', '+2.852', '+2.753', '+2.815', '+2.730'], ['+3.790', '+3.367', '+3.090', '+2.840', '+2.715'], ['+3.870', '+3.770', '+3.955', '+4.000', '+3.857']]
[2021-01-12 14:25:06,858] DEBUG: mesh collection started
[2021-01-12 14:27:54,072] DEBUG: Canceling mesh collection per user request
[2021-01-12 14:27:54,073] DEBUG: Mesh data collected prior to cancel:
[2021-01-12 14:27:54,073] DEBUG: []
[2021-01-12 14:27:54,073] DEBUG: Mesh data after clearing:
[2021-01-12 14:27:54,073] DEBUG: []
[2021-01-12 14:35:09,682] DEBUG: mesh collection started
[2021-01-12 14:38:49,975] DEBUG: Canceling mesh collection per user request
[2021-01-12 14:38:49,975] DEBUG: Mesh data collected prior to cancel:
[2021-01-12 14:38:49,976] DEBUG: []
[2021-01-12 14:38:49,976] DEBUG: Mesh data after clearing:
[2021-01-12 14:38:49,976] DEBUG: []

**Screenshots**
Current settings
[image]

**Firmware and Version**
Octoprint: Version 1.5.2
printer.firmware: Marlin 2.0.7.2 (Oct 16 2020)
Hardware: RasPI 4

jneilliii, 2021-01-12:
How big is your probing mesh? I see where it returned a 5x5 grid at some point in your debug log. Is the full probing process able to complete prior to the timeout of 1800 secs? What type of bed leveling does your firmware use?

lopio1234, 2021-01-12:
> 
> 
> How big is your probing mesh? I see where it returned a 5x5 grid at some point in your debug log. Is the full probing process able to complete prior to the timeout of 1800 secs? What type of bed leveling does your firmware use?

I hope i got your questtions right.

1. In the bed level firmware setteings are 25 probing points on a 220 x 220 bed
2. Am i right that 1800 secs mean 30 minutes then yes, normally the bed leveling took 1 - 4 minutes 
3. For the leveling a use a bltouch with the pre configured firmware Marlin 2.0.7.2

i hope i got all questions right for you

jneilliii, 2021-01-12:
Thanks, so that is what I see in the log. A 5x5 grid = 25 points. 

>[2021-01-12 14:19:40,168] DEBUG: ['+0.312', '+0.217', '+0.102', '-0.015', '-0.125']
[2021-01-12 14:19:40,172] DEBUG: ['+2.955', '+2.855', '+0.200', '+0.102', '-0.160']
[2021-01-12 14:19:40,174] DEBUG: ['+2.943', '+2.852', '+2.753', '+2.815', '+2.730']
[2021-01-12 14:19:40,176] DEBUG: ['+3.790', '+3.367', '+3.090', '+2.840', '+2.715']
[2021-01-12 14:19:40,179] DEBUG: ['+3.870', '+3.770', '+3.955', '+4.000', '+3.857']

Some of the numbers are extremely high, and therefore outside the graphing z limits. I would recommend that you do a round of manual leveling first using the paper drag technique and then run your probe again and see if that makes any difference.

closed 2021-02-08 by github-actions[bot]