jneilliii/OctoPrint-BedLevelVisualizer #358: [FR]: Clear description of which Y direction is assumed

defl, 2021-01-17:
When trying to configure my Ender3Pro (running Klipper) I was confused about the visualization of the bed.

For testing I've put a credit card sized object on the bed, top left corner (x low, y high). klipper does a zig-zag pattern when probing points and from that I get:

> Send: BED_MESH_CALIBRATE
> Recv: // probe at 57.000,15.000 is z=0.514000
> Recv: // probe at 57.000,15.000 is z=0.519000
> Recv: // probe at 89.163,15.000 is z=0.484000
> Recv: // probe at 89.163,15.000 is z=0.494000
> Recv: // probe at 121.325,15.000 is z=0.456500
> Recv: // probe at 121.325,15.000 is z=0.459000
> ...
> Recv: // probe at 217.800,181.650 is z=0.611500
> Recv: // probe at 185.637,181.650 is z=0.584000
> Recv: // probe at 185.637,181.650 is z=0.589000
> Recv: // probe at 153.475,181.650 is z=0.559000
> Recv: // probe at 153.475,181.650 is z=0.581500
> Recv: // probe at 121.325,181.650 is z=1.449000 <-- onto the credit card (from right to left)
> Recv: // probe at 121.325,181.650 is z=1.461500
> Recv: // probe at 89.163,181.650 is z=1.594000
> Recv: // probe at 89.163,181.650 is z=1.619000
> ...
> Recv: // probe at 57.000,214.975 is z=1.724000
> Recv: // probe at 89.163,214.975 is z=1.694000
> Recv: // probe at 89.163,214.975 is z=1.694000
> Recv: // probe at 121.325,214.975 is z=1.556500
> Recv: // probe at 121.325,214.975 is z=1.554000
> Recv: // probe at 153.475,214.975 is z=0.659000 <-- off of the credit card (from left to right)
> Recv: // probe at 153.475,214.975 is z=0.639000
> Recv: // probe at 185.637,214.975 is z=0.656500

So far so good. bed mesh output

> Send: BED_MESH_OUTPUT
> Recv: // Mesh Leveling Probed Z positions:
> Recv: // 0.017500 -0.010000 -0.041250 0.021250 0.070000 0.176250 0.297500
> Recv: // -0.015000 -0.053750 -0.091250 -0.050000 -0.035000 0.051250 0.171250
> Recv: // 0.031250 -0.003750 -0.056250 -0.010000 0.015000 0.081250 0.161250
> Recv: // 0.083750 0.028750 -0.023750 0.015000 0.001250 0.050000 0.136250
> Recv: // 0.128750 0.073750 0.016250 0.037500 0.036250 0.095000 0.170000
> Recv: // 1.093750 1.107500 0.956250 0.071250 0.087500 0.117500 0.200000
> Recv: // 1.228750 1.195000 1.056250 0.150000 0.147500 0.203750 0.298750
> 

Here you can see that the higher y is lower in the table (so inverted from human perception). So far also so good. However then we get to the output which is used to plot the bed (note that there are 2 interpolation points between every measurement):

> Recv: Measured points:
> Recv:   1.228750  1.221556  1.208361  1.195000  1.169398  1.120630  1.056250  0.803796  0.396398  0.150000  0.121630  0.133093  0.147500  0.161435  0.185481  0.203750  0.231454  0.273546  0.298750
> Recv:   1.206046  1.202770  1.195990  1.186333  1.159940  1.108901  1.042769  0.788009  0.377167  0.128917  0.101913  0.116041  0.131815  0.143871  0.164443  0.180444  0.207014  0.247722  0.272130
> Recv:   1.159343  1.162333  1.165815  1.162111  1.134512  1.080017  1.011509  0.755364  0.342769  0.093833  0.069292  0.087604  0.105463  0.114492  0.129465  0.141806  0.166493  0.204881  0.227954
> Recv:   1.093750  1.099759  1.108213  1.107500  1.079565  1.024130  0.956250  0.708972  0.311083  0.071250  0.049037  0.069046  0.087500  0.094981  0.107074  0.117500  0.141000  0.177833  0.200000
> Recv:   0.825935  0.826792  0.826993  0.822250  0.797033  0.749581  0.696250  0.517956  0.232152  0.060000  0.043938  0.058323  0.072194  0.081581  0.097048  0.109444  0.132449  0.168003  0.189352
> Recv:   0.392565  0.384085  0.369303  0.357111  0.336536  0.302687  0.273583  0.208740  0.107194  0.046250  0.039912  0.044422  0.050444  0.063352  0.085100  0.101222  0.123704  0.157535  0.177759
> Recv:   0.128750  0.114528  0.090528  0.073750  0.056046  0.030565  0.016250  0.020389  0.030861  0.037500  0.036917  0.035167  0.036250  0.051204  0.076657  0.095000  0.117296  0.150315  0.170000
> Recv:   0.088602  0.073679  0.048563  0.031269  0.014165  -0.009874  -0.022083  -0.009111  0.015617  0.030704  0.029848  0.025621  0.024935  0.039122  0.063629  0.081537  0.104579  0.138964  0.159491
> Recv:   0.083343  0.068604  0.043802  0.026731  0.009972  -0.013602  -0.025750  -0.014207  0.008053  0.021407  0.018828  0.012067  0.009676  0.022182  0.044347  0.061074  0.085411  0.122262  0.144315
> Recv:   0.083750  0.069454  0.045380  0.028750  0.012157  -0.011324  -0.023750  -0.014481  0.004009  0.015000  0.011657  0.004148  0.001250  0.012926  0.033880  0.050000  0.075083  0.113333  0.136250
> Recv:   0.068713  0.055935  0.034364  0.019250  0.002762  -0.020874  -0.033324  -0.023486  -0.003952  0.008074  0.007598  0.004685  0.004722  0.017888  0.040616  0.057676  0.082343  0.119640  0.141954
> Recv:   0.046343  0.036061  0.018603  0.005972  -0.010375  -0.034327  -0.046898  -0.036455  -0.015733  -0.002296  0.002097  0.007085  0.012194  0.028068  0.054091  0.072907  0.096823  0.132385  0.153602
> Recv:   0.031250  0.022435  0.007398  -0.003750  -0.019861  -0.043750  -0.056250  -0.045500  -0.024194  -0.010000  -0.002759  0.006870  0.015000  0.032713  0.061093  0.081250  0.105139  0.140306  0.161250
> Recv:   0.016537  0.007498  -0.007871  -0.019065  -0.034071  -0.056113  -0.067546  -0.057047  -0.036431  -0.022759  -0.016220  -0.007599  0.000148  0.019203  0.050031  0.072102  0.099109  0.139052  0.162861
> Recv:   -0.004065  -0.013794  -0.030237  -0.041824  -0.055022  -0.073972  -0.083620  -0.073576  -0.054190  -0.041519  -0.036717  -0.030709  -0.024204  -0.003259  0.031216  0.056231  0.088463  0.136482  0.165139
> Recv:   -0.015000  -0.025065  -0.042019  -0.053750  -0.065787  -0.082769  -0.091250  -0.081278  -0.062306  -0.050000  -0.045944  -0.041056  -0.035000  -0.012694  0.024278  0.051250  0.086694  0.139639  0.171250
> Recv:   -0.007463  -0.016745  -0.032388  -0.043241  -0.054683  -0.070817  -0.078583  -0.067124  -0.045752  -0.031657  -0.025070  -0.016309  -0.007704  0.016509  0.056071  0.084620  0.120664  0.174208  0.206148
> Recv:   0.007907  -0.000005  -0.013364  -0.022731  -0.033451  -0.048630  -0.055472  -0.041512  -0.016034  0.001130  0.012095  0.027540  0.040481  0.067547  0.110830  0.141546  0.177841  0.231244  0.263046
> Recv:   0.017500  0.010426  -0.001537  -0.010000  -0.020306  -0.034944  -0.041250  -0.025769  0.002213  0.021250  0.034889  0.054417  0.070000  0.098769  0.144259  0.176250  0.212630  0.265843  0.297500

There he output is inverted from the machine stored output (so human view, with earlier lines being higher x). Using this out of the box mis renders the bump to be bottom left, not bottom right.

My visualizer config is as follows. Key thing to note is that I need to "flip y axis".
[image]

And that leads to the following image (which is correct)
[image]

Now, what I can gather from the Klipper developers is that they do this inverse ordering because of you guys. 
https://github.com/KevinOConnor/klipper/issues/2490:

"@arksine commented on 8 Feb 2020: Indeed, the interpolated values in BED_MESH_OUTPUT are inverted as if you were looking down at the bed, with the origin in the bottom left corner. It was done this way to retain compatibility with other firmware. When I originally wrote bed_mesh both Prusa Firmware and Marlin also output the mesh in this manner. Some visualizer plugins use the output to generate height maps."

Now it seems that the reason why this is inverted (and confusing) needs further config to unconfuse it. So either we need to rename the "flip y axis" to  something that describes the expected inputs like "y=0 is bottom" and "Y=0 is top". Even better would be some documentation on what you expect as an input so Klipper and co can start to flip their ordering.

jneilliii, 2021-01-17:
Thank you for such a detailed explanation of your trouble. This is in fact why those flip and rotate options are there, because not every firmware reports the same way and provided in the same format. The way the plugin works is once the triggering flag (@BEDLEVELVISUALIZER) is seen it starts analyzing every line received from the printer looking for mesh data. If it detects it as mesh data it will then append/push that data to an internal array until the plugin receives an ok from the printer; confirming the collection of data. There is then other logic that will manipulate that array once collected based on configured options and other variations created from different firmware reports, etc. and send that collected data to the web interface for graphing. 

Your approach for determining the bed orientation is spot on as well, using an obstacle to trigger earlier in order to show higher. I like the idea of changing the options to "y=0 is bottom" and "y=0 is top" but I fear that would also confuse people as much as the "flip Y" checkbox is now and depending on the firmware's mesh report may actually not match that text.

What I could do however is add some additional information to the readme/gcode examples to give user's a heads up that if they are using Klipper they would want to enable that option. Long term goals for this plugin is utilize firmware detection and just have a lot of this stuff pre-configured for you, but I haven got much time these days to work on this stuff.

jneilliii, 2021-01-17:
This might also be covered by #263, which crosslink said he was going to do a YouTube video but has still not come through yet.

defl, 2021-01-17:
Thanks for the quick reply! 👍 

In the end I think this plugin is one of the most used plugins for 3d bed leveling. (The other main one (AFAIK) is Duet where the firm controls both the hw and the software so this never occurs). You can see that by firmware designers copying what others are doing just to make sure this plugin doesn't break. IMHO an argument can be made for either direction (lower lines are higher/lower Y) so trying to set a standard based on that won't get far I guess... In stead of you trying to guess firmware who are guessing at what you want, how about you add support for firmware to tell you their direction? (If that doesn't already exist in some way, did a quick search but could not find it) Sure it'll take a ton of time to get it in everywhere but eventually we'll get there.

jwvanderbeck, 2021-01-27:
I've literally been spending hours messing with this and am so confused on what I am actually seeing, which corner is which, etc.  I know it is probably a lot of work but having presets for known setups might be helpful to people.  There are probably only a few **main** setups out there.

That credit card trick is gold though! Going to do that right now and maybe I'll finally understand what I'm seeing.

bratwiz, 2021-11-12:
> That credit card trick is gold though! Going to do that right now and maybe I'll finally understand what I'm seeing.

I totally agree. Thanks for the tip!!

PR #497 (2022-01-01): 1.1.1
* add ability to configure graph height in visualization settings, resolves #462
* add min, max, and variance values of graphed mesh, resolves #286
* add OctoDash support, create Custom Action with `[!WEB]http://127.0.0.1:5000/plugin/bedlevelvisualizer/bedlevelvisualizer` as the command.
* adjust the tables for better theme support, resolves #479
* add option for showing mesh data on tab, resolves #496
* remove tooltip hover on tables, resolves #490
* add Prusa adjustment values to graph (still needs verification), resolves #242
* add appearance > title setting to automatically downloade [... cut]
closed 2022-01-01 by jneilliii