jneilliii/OctoPrint-BedLevelVisualizer #359: [BUG]: Bed Level visualizer not "installing"

nfons, 2021-01-19:
**Describe the bug**

when installing the app, it "installs"

But does not seem to actually register with octopi:

`Installing plugin "Bed Level Visualizer" from https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip.../home/pi/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpyw9e43k_/OctoPrint-BedLevelVisualizer-master.zip --no-cache-dirLooking in indexes: https://pypi.org/simple, https://www.piwheels.org/simpleProcessing /tmp/tmpyw9e43k_/OctoPrint-BedLevelVisualizer-master.zipRequirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from Bed-Visualizer==1.0.0) (1.5.3)Requirement already satisfied: numpy<=1.19.2,>=1.16.0 in /home/pi/oprint/lib/python3.7/site-packages (from Bed-Visualizer==1.0.0) (1.19.2)Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (5.3.1)Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (0.1.5.post0)Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (0.17.8)Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (1.2.0)Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (0.4.21)Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (2020.8.7)Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (2.8.5)Requirement already satisfied: zeroconf<0.25,>=0.24; python_version >= "3" in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (0.24.5)Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (1.1.2)Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (1.4)Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (3.1.1)Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (0.1.1)Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (5.1.1)Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (1.1.0)Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (1.1.1)Requirement already satisfied: watchdog==0.10.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (0.10.4)Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (0.5.0)Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (1.12.1)Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (4.0)Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (1.0.0)Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (5.7.2)Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (1.2)Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (2.24.0)Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (3.4)Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (0.8.0)Requirement already satisfied: feedparser<7,>=6.0.2; python_version >= "3" in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (6.0.2)Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (7.1.2)Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (1.0.7)Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (0.10.9)Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (2.0)Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (0.57.0)Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (1.0.1)Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (2020.7.14)Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (2.11.2)Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (0.6.0)Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (1.5.0.1)Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (0.18.2)Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==1.0.0) (2020.9.23)Requirement already satisfied: certifi in /home/pi/oprint/lib/python3.7/site-packages (from sentry-sdk<1,>=0.15.1->OctoPrint->Bed-Visualizer==1.0.0) (2020.6.20)Requirement already satisfied: urllib3>=1.10.0 in /home/pi/oprint/lib/python3.7/site-packages (from sentry-sdk<1,>=0.15.1->OctoPrint->Bed-Visualizer==1.0.0) (1.25.10)Requirement already satisfied: ifaddr in /home/pi/oprint/lib/python3.7/site-packages (from zeroconf<0.25,>=0.24; python_version >= "3"->OctoPrint->Bed-Visualizer==1.0.0) (0.1.7)Requirement already satisfied: setuptools>=36 in /home/pi/oprint/lib/python3.7/site-packages (from markdown<3.2,>=3.1->OctoPrint->Bed-Visualizer==1.0.0) (50.3.0)Requirement already satisfied: pathtools>=0.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from watchdog==0.10.4->OctoPrint->Bed-Visualizer==1.0.0) (0.1.2)Requirement already satisfied: pyasn1>=0.1.3 in /home/pi/oprint/lib/python3.7/site-packages (from rsa==4.0->OctoPrint->Bed-Visualizer==1.0.0) (0.4.8)Requirement already satisfied: Babel>=2.3 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->Bed-Visualizer==1.0.0) (2.8.0)Requirement already satisfied: pytz in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->Bed-Visualizer==1.0.0) (2020.1)Requirement already satisfied: idna<3,>=2.5 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.23.0->OctoPrint->Bed-Visualizer==1.0.0) (2.10)Requirement already satisfied: chardet<4,>=3.0.2 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.23.0->OctoPrint->Bed-Visualizer==1.0.0) (3.0.4)Requirement already satisfied: sgmllib3k in /home/pi/oprint/lib/python3.7/site-packages (from feedparser<7,>=6.0.2; python_version >= "3"->OctoPrint->Bed-Visualizer==1.0.0) (1.0.0)Requirement already satisfied: webassets>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Assets<3,>=2.0->OctoPrint->Bed-Visualizer==1.0.0) (2.0)Requirement already satisfied: six in /home/pi/oprint/lib/python3.7/site-packages (from websocket-client<1,>=0.57->OctoPrint->Bed-Visualizer==1.0.0) (1.15.0)Building wheels for collected packages: Bed- [... cut ...] uilt Bed-VisualizerInstalling collected packages: Bed-VisualizerAttempting uninstall: Bed-VisualizerFound existing installation: Bed-Visualizer 1.0.0Uninstalling Bed-Visualizer-1.0.0:Successfully uninstalled Bed-Visualizer-1.0.0Successfully installed Bed-Visualizer-1.0.0Done!`

in the restart popup, we see the name being unknown:

[image]

once restarted, the bed level is not in the installed plugin list

**Expected behavior**

**Debug Log**

**Screenshots**

**Firmware and Version**
latest 1.5+

jneilliii, 2021-01-19:
Python 3? If so read known issues in the readme for a fix.

nicojako, 2021-01-19:
Have the same Problem.
[image]

jneilliii, 2021-01-20:
Probably the same will fix it for you... @nicojako  see the first item listed here.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer#known-issues

TheMelvinNL, 2021-01-23:
Same issue, even after installing libatlas3-base

nicojako, 2021-01-23:
Install python 3

TheMelvinNL, 2021-01-23:
Can I just install python 3 via the SSH through apt? Do I have to configure anything else?

jneilliii, 2021-01-23:
Easiest way to install python 3 is to use the upgrade script. https://github.com/cp2004/Octoprint-Upgrade-To-Py3

nfons, 2021-02-01:
This worked with the update script @jneilliii posted

closed 2021-02-01 by nfons

o-nix, 2021-02-17:
Seems my board has Wi-Fi connection issues, so `numpy` installation was broken. In my case, this was leading to repeated "silent" plugin failures after "successful" (as `numpy` was, technically, present) installation reports.
I did `~/oprint/bin/pip uninstall numpy bed-visualizer`, restarted, and reinstalled the plugin via OctoPrint's manager.