jneilliii/OctoPrint-BedLevelVisualizer #362: [BUG]: every time I hit update mesh it goes to home touches twice and just sits there

midget1960, 2021-01-22:
**Describe the bug**

**Expected behavior**

**Debug Log**

**Screenshots**

**Firmware and Version**

jneilliii, 2021-01-22:
I can't help you unless you fill in the template...

midget1960, 2021-01-22:
Sorry I have bltouch v 3.1 using octoprint I have 
G29 
M155 S30
@BEDLEVELVISUALIZER
G29 T
M155 S3
And when I hit update mesh it goes to home touches twice and just sits there it won’t continue on and do the bed level I tried clearing data and rebooting octoprint and it still does the same thing when I go to my ender 3 pro with the skr mini v1.2 board and hit level bed it go’s through the hole process just fine.

midget1960, 2021-01-22:
I have it figured out so it’s working thanks

jneilliii, 2021-01-22:
Yeah, you had the flag in the wrong spot. Glad you got it worked out.

closed 2021-01-22 by jneilliii