jneilliii/OctoPrint-BedLevelVisualizer #364: [BUG]: Mesh not updating

Daskew78, 2021-01-23:
I'm having a problem with the Bed Visualizer plugin in that no matter what adjustments I make to the bed it's always showing the same graph after re-probing the bed/updating the mesh.

I've tried the following:
Clear up Plugin Data
Restart Octoprint
Restart Raspberry Pi
Re-install the plugin and restart Raspberry Pi
Restart the printer
A different browser

It would appear measure plot points have become stuck and just refuse to be overwritten with new values.
I do come across this now and again on different machines but normally clearing the plugin data and/or re-installing the plugin resolves it but on this occasion it will update.

**Expected behavior**
When updating the mesh with an adjusted bed an updated mesh graph should appear.

**Debug Log**
This is 2 different probes after adjusting all 4 wheels quite dramatically and the graph appears identical:
[plugin_bedlevelvisualizer_debug (1).log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/5860669/plugin_bedlevelvisualizer_debug.1.log)

**Screenshots**
[image]

**Firmware and Version**
Marlin bugfix-2.0.x

jneilliii, 2021-01-23:
That's odd. I do see the collected meshes as being different. Do you have any errors in your browser's developer console (F12 on your keyboard and make sure console tab is selected).

Daskew78, 2021-01-24:
Not that I can see, this is the console:
Starting dependency resolution...
packed_core.js?b8ae0794:1003 ... dependency resolution done
packed_core.js?b8ae0794:1026 Initial application setup done, connecting to server...
packed_core.js?b8ae0794:838 Connected to the server
packed_core.js?b8ae0794:843 Triggering connect on all view models
packed_core.js?b8ae0794:845 Server (re)connect processed
packed_core.js?b8ae0794:1026 Finalizing application startup
packed_core.js?b8ae0794:1007 Going to bind 57 view models...
packed_plugins.js?23f8eb4a:20836 octolapse.help.js - Binding help links to #octolapse_tab a.octolapse_help[data-help-url]
packed_core.js?b8ae0794:1019 Did not bind view model TimelapseViewModel to target #timelapse since it does not exist
packed_core.js?b8ae0794:1019 Did not bind view model TimelapseViewModel to target #timelapse_link since it does not exist
packed_core.js?b8ae0794:1019 Did not bind view model BackupViewModel to target #wizard_plugin_backup since it does not exist
packed_core.js?b8ae0794:1019 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?b8ae0794:1019 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?b8ae0794:1019 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?b8ae0794:1019 Did not bind view model BLTouchViewModel to target settings_plugin_BLTouch_form since it does not exist
packed_core.js?b8ae0794:1019 Did not bind view model bedlevelvisualizerViewModel to target #wizard_plugin_bedlevelvisualizer since it does not exist
packed_core.js?b8ae0794:1019 Did not bind view model WebcamTabViewModel to target #webcamtab since it does not exist
packed_plugins.js?23f8eb4a:10013 onBeforeBinding: settings= Object
packed_core.js?b8ae0794:159 User Daskew78 logged in
packed_core.js?b8ae0794:1024 ... binding done
packed_core.js?b8ae0794:1024 Application startup complete
packed_plugins.js?23f8eb4a:7306 Array(5)
packed_plugins.js?23f8eb4a:7441 Back [um]:-61
packed_plugins.js?23f8eb4a:7442 Front [um]:64
packed_plugins.js?23f8eb4a:7443 Left [um]:1
packed_plugins.js?23f8eb4a:7444 Right [um]:9
packed_plugins.js?23f8eb4a:10266 ExcludeRegionPlugin:: excludedRegions updated, redrawing GCODE viewer: excludedRegions= Array(0)
2packed_plugins.js?23f8eb4a:20836 octolapse.help.js - Binding help links to .octolapse .webcam_settings a.octolapse_help[data-help-url]
6The resource <URL> was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/sockjs/info?t=1611445255922:1 Failed to load resource: net::ERR_CONNECTION_REFUSED
/sockjs/info?t=1611445258987:1 Failed to load resource: net::ERR_CONNECTION_REFUSED
/sockjs/info?t=1611445274338:1 Failed to load resource: net::ERR_CONNECTION_TIMED_OUT
/sockjs/info?t=1611445298339:1 Failed to load resource: the server responded with a status of 502 (Bad Gateway)
packed_core.js?b8ae0794:838 Connected to the server
packed_core.js?b8ae0794:159 User needs for Daskew78 changed
packed_core.js?b8ae0794:843 Triggering reconnect on all view models
packed_core.js?b8ae0794:845 Server (re)connect processed
packed_plugins.js?23f8eb4a:10266 ExcludeRegionPlugin:: excludedRegions updated, redrawing GCODE viewer: excludedRegions= Array(0)
2packed_plugins.js?23f8eb4a:20836 octolapse.help.js - Binding help links to .octolapse .webcam_settings a.octolapse_help[data-help-url]
packed_plugins.js?23f8eb4a:7306 Array(5)
packed_plugins.js?23f8eb4a:7441 Back [um]:-61
packed_plugins.js?23f8eb4a:7442 Front [um]:64
packed_plugins.js?23f8eb4a:7443 Left [um]:1
packed_plugins.js?23f8eb4a:7444 Right [um]:9
packed_plugins.js?23f8eb4a:7488 Array(6)
packed_plugins.js?23f8eb4a:7306 Array(5)
packed_core.js?b8ae0794:658 ConfiguredLoggers has not changed. Not saving.
packed_plugins.js?23f8eb4a:7441 Back [um]:127
packed_plugins.js?23f8eb4a:7442 Front [um]:-79
packed_plugins.js?23f8eb4a:7443 Left [um]:57
packed_plugins.js?23f8eb4a:7444 Right [um]:-47
packed_plugins.js?23f8eb4a:7306 Array(5)
packed_plugins.js?23f8eb4a:7441 Back [um]:127
packed_plugins.js?23f8eb4a:7442 Front [um]:-79
packed_plugins.js?23f8eb4a:7443 Left [um]:57
packed_plugins.js?23f8eb4a:7444 Right [um]:-47
packed_plugins.js?23f8eb4a:7488 Array(6)
packed_plugins.js?23f8eb4a:7306 Array(5)
packed_core.js?b8ae0794:658 ConfiguredLoggers has not changed. Not saving.
packed_plugins.js?23f8eb4a:7441 Back [um]:175
packed_plugins.js?23f8eb4a:7442 Front [um]:-107
packed_plugins.js?23f8eb4a:7443 Left [um]:86
packed_plugins.js?23f8eb4a:7444 Right [um]:-73
packed_plugins.js?23f8eb4a:7488 Array(6)
packed_plugins.js?23f8eb4a:7306 Array(5)
packed_core.js?b8ae0794:658 ConfiguredLoggers has not changed. Not saving.
packed_plugins.js?23f8eb4a:7441 Back [um]:155
packed_plugins.js?23f8eb4a:7442 Front [um]:-93
packed_plugins.js?23f8eb4a:7443 Left [um]:76
packed_plugins.js?23f8eb4a:7444 Right [um]:-67
packed_plugins.js?23f8eb4a:7488 Array(6)
packed_plugins.js?23f8eb4a:7306 Array(5)
packed_core.js?b8ae0794:658 ConfiguredLoggers has not changed. Not saving.
packed_plugins.js?23f8eb4a:7441 Back [um]:129
packed_plugins.js?23f8eb4a:7442 Front [um]:-81
packed_plugins.js?23f8eb4a:7443 Left [um]:64
packed_plugins.js?23f8eb4a:7444 Right [um]:-65

Daskew78, 2021-01-24:
Actually when I pasted it in here it's easier to read and there does look to be a few errors in there

jneilliii, 2021-01-24:
> packed_plugins.js?23f8eb4a:7441 Back [um]:129
packed_plugins.js?23f8eb4a:7442 Front [um]:-81
packed_plugins.js?23f8eb4a:7443 Left [um]:64
packed_plugins.js?23f8eb4a:7444 Right [um]:-65

ok, this is from my plugin. It's not an error but more of a future feature that I've been working on. I don't see any other errors in here. Do you have any other plugins installed that also involve graphing similar to this one?

Daskew78, 2021-01-24:
No, nothing similar to this one, these are the plugins:
Plugins installed
- DisplayLayerProgress Plugin
- GcodeEditor
- HeaterTimeout
- Exclude Region
- BLTouch Plugin
- OctoPod Plugin
- Cancel Objects
- Tab Order
- Themeify
- Fan Speed Control
- Simple Emergency Stop
- WebcamTab
- Bed Visualizer
- Detailed Progress
- Extra Distance Buttons
- Navbar Temperature Plugin
- Creality-2x-temperature-reporting-fix
- OctoPrint Anywhere
- Dashboard
- Floating Navbar
- PrintTimeGenius Plugin
- Octolapse
- 
I'm doing more testing as we speak but starting to think it might be the bltouch as I've reverted to an old backup of the SD card where it was working and it's still happening. But like you say the values are different so it can't be. Even my mobile phone shows the same graph so that rules out my PC as well.

jneilliii, 2021-01-24:
Is the current mesh data or angle adjustments in the plugin's settings updating after using the update mesh button?

Daskew78, 2021-01-24:
Yes it's updating, but I'm getting somewhere with this now...

I've swapped it to a Pi that I was using on another printer and it's working now :/

It must be the Pi in this case and i'll do a re-install.... would you agree?

jneilliii, 2021-01-24:
maybe re-flash the sd card with octopi 0.18rc2 32 bit version.

Daskew78, 2021-01-24:
Noted, thanks for the super support and fast replies :)

closed 2021-02-14 by github-actions[bot]