jneilliii/OctoPrint-BedLevelVisualizer #368: Leveling Knob Rotation Algorithm

beezischillin, 2021-01-26:
Dear author,
First of all, I would like to thank you for this plugin, it is such an amazing and helpful tool, especially as a newbie to printing! I have a question I would like your help with, however. Can you please help me out by providing me the algorithm I could use to calculate how much any leveling screw needs to be rotated to level the other 3 extremities of the bed compared to the first point? I'm not so well-versed in python so I'm having a bit of a hard time figuring it out.

I thought it would be a fun project to attempt to automate the process while trying to get results as precise as possible using an arduino and 4 light-ish steppers.

Thank you and sorry for the bother!

jneilliii, 2021-01-26:
That math was actually done by someone else that I merged into the plugin. I think all of it is done in the js dude though, not python, as part of the knockout binding in the jinja templates.

beezischillin, 2021-01-26:
> That math was actually done by someone else that I merged into the plugin. I think all of it is done in the js dude though, not python, as part of the knockout binding in the jinja templates.

Thank you for the prompt response and the help! I will look through the JS and close this so it doesn't clutter up your issues!

closed 2021-01-26 by beezischillin

jneilliii, 2021-01-26:
Sure, it's actually in the knockout binding side (which is also js code kinda). It starts here.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/4ccbac48fc33792a6d6fa473ddd5c1a659ab65ff/octoprint_bedlevelvisualizer/templates/bedlevelvisualizer_tab.jinja2#L68

beezischillin, 2021-01-26:
Yes! Found it, thank you!