jneilliii/OctoPrint-BedLevelVisualizer #369: Suddenly Timing Out

deputyd0ng, 2021-01-27:
I had this running just fine for months on my CR-10s.  I started getting timeout errors recently.  I tried increasing the interval, that did not resolve.  

I removed+reinstalled, and when that didnt work I formatted my Pi SD and installed it on a fresh version of octoprint with no other plug ins running, same thing, bed probes, but it doesnt display results, and times out.

Ive enabled debug logging, where are the logs stored?

jneilliii, 2021-01-27:
Inside the logging section of OctoPrint's settings, file named plugin_belevelvisualizer_debug.log. Share your octoprint.log too if you want me to look.

deputyd0ng, 2021-01-28:
[octoprintlogs.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/5883947/octoprintlogs.log)
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/5883948/plugin_bedlevelvisualizer_debug.log)

jneilliii, 2021-01-28:
Yeah, nothing showing up directly in any of those that might indicate what's happening. You are running an older version of OctoPrint it seems, and there was a known bug at one point where settings could get wiped out that would cause weird issues. If you check your config.yaml file do you see the word null under any of the settings in plugins > bedlevelvisualizer? From what I can tell from the BLV log, it's not getting any info back from your printer.

jneilliii, 2021-01-28:
Oh, wait I see that this was a fresh flash of octopi 0,17 now and was upgraded to 1.5.3. Either way, the plugin doesn't see the firmware saying anything to the gcode commands you have configured to interpret the message.

deputyd0ng, 2021-01-28:
I'm using the stock commands for bilinear bed leveling. I'll check the yaml and report back.

deputyd0ng, 2021-01-28:
bedlevelvisualizer:
        _config_version: 1
        command: "G28      ; aome all axes\nM155 S30 ; reduce temperature reporting\
            \ rate to reduce output pollution\n@BEDLEVELVISUALIZER\t; tell the plugin\
            \ to watch for reported mesh\nG29 T \t   ; run bilinear probing\nM155\
            \ S3  ; reset temperature reporting"
        debug_logging: true
        timeout: '200'

deputyd0ng, 2021-01-28:
Also, when  checking terminal after probing I see this:

Recv:       0      1      2      3      4
Recv:  0 -11.495 -11.500 -11.540 -11.613 -11.740
Recv:  1 -11.065 -11.153 -11.222 -11.375 -11.635
Recv:  2 -11.020 -11.097 -11.140 -11.220 -11.403
Recv:  3 -11.055 -11.155 -11.200 -11.265 -11.417
Recv:  4 -11.212 -11.300 -11.352 -11.403 -11.477

deputyd0ng, 2021-01-28:
I went through my firmware again in VS, and reflashed, just to be sure, still no dice.

jneilliii, 2021-01-28:
Yes, it appears something got a little messed up in your settings possibly. It might be something I need to look into fixing. But as a test can you try to delete the gcode settings completely and then type them in one line at a time to see if that fixes it? I would just put in the commands, not the comments, like this.

```
G28
M155 S30
@BEDLEVELVISUALIZER
G29 T
M155 S3
```

deputyd0ng, 2021-01-28:
Just did that no good, must be the firmware.

Any idea where to start looking in Marlin?

jneilliii, 2021-01-28:
Oh, I see the issue now. This is actually fixed in my 1.0.1rc2 version right now. Your probe points are in the higher numbers and there was a parsing error for that. If you go into OctoPrint's Software Update Settings and switch the plugin to use Release Candidate in the pull down you should get an update notice to install that version.

jneilliii, 2021-01-28:
duplicate of #352

jneilliii, 2021-01-28:
please post in above linked issue if the rc doesn't resolve this for you.

closed 2021-01-28 by jneilliii