jneilliii/OctoPrint-BedLevelVisualizer #370: [FR]: Supporting more then 4x4 probing grid

theboy0007, 2021-01-28:
Is it possible to make the plugin support more then 4x4 grid probing? I am running a 10x10 grid probing on my ender 5 but if i want to use the plugin to visualize this, the serial connection from octoprint to my printer disconnects.

jneilliii, 2021-01-28:
Have you considered increasing your timeouts in OctoPrint and/or verified that your G29 command is in the Long Running Command section of Firmware & protocol > Advanced Options? 

[image]

I can't control what OctoPrint does relative to timeout and the plugin supports as many points that can be reported by the firmware. Depending on what type of leveling you are using you could potentially utilize the stored EEPROM mesh data instead by using the `M420 V` command instead of a full G29 probing, then do the probe outside of OctoPrint.

KenLucke, 2021-02-01:
You can also speed up the probe itself, which may help, by specifying a speed.  I use 
G29 T S65000

closed 2021-02-23 by github-actions[bot]