jneilliii/OctoPrint-BedLevelVisualizer #374: [FR]: Mousing over bed highlights point in chart and  corresponding wheel turn angle

KenLucke, 2021-02-01:
**Is your feature request related to a problem? Please describe.**
no

**Describe the solution you'd like**
It would be slick if when you are mousing over the graph visualization, the corresponding details in the chart below highlighted, and the corresponding wheel angle showed.

**Describe alternatives you've considered**

**Additional context**

jneilliii, 2022-07-16:
Added with the above commit to the rc branch and will be available in next rc release after getting a couple of more issues completed this weekend. You can change the plugin's release channel in OctoPrint's Software Update settings to Release Candidate and update when it comes out, or manually install by copying/pasting the URL below into plugin manager > get more > ...from URL and clicking install.

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/refs/heads/rc.zip
```

closed 2022-07-16 by jneilliii

jneilliii, 2022-07-17:
I'm thinking however after making some additional changes to the hover markers I might be getting rid of the adjustment table completely. Kind of redundant.

[image]