jneilliii/OctoPrint-BedLevelVisualizer #377: Rotate Bed visualization

screechyboy79, 2021-02-03:
I might be being a bit dense but after running the bed visualization the graphic that appears in OctoPrint shows the front as Y0 X0 where as in fact on my Ender 5 its Y200 X200 meaning with each bed visualisation I run I have to rotate the visualisation or I end up adjusting the wrong side of bed.

I have changed the Clockwise Rotation setting to 90 and 180 but still the front of the graphic is Y0X0.

jneilliii, 2021-02-03:
Yeah, this is something that's been on my todo for a little bit. If you use the rotation options it just rotates the data itself, not the grid in the visualization. In theory if you use the rotate option it will rotate the mesh properly, but will still always graph 0,0 in the front left corner, because the majority of firmware/printers out there this physical orientation is correct and front left corner is 0,0.

screechyboy79, 2021-02-03:
Thank you for replying so quickly... good to know it does rotate the data even if the axis do not change I got myself in a muddle earlier but at least now I know I can ignore the axis label. Its a fantastic plugin so thank you very much for all your work on it!

tarsonis123, 2021-04-23:
Hi there. This is the topic I was trying to find something about in the docs. Since I'm using Klipper I'm confused if 0;0 is really showing me the data for the font left. I'm missing the commands I used to have in Marlin where you can send the toolhead to position with the mesh point number as command. With Klipper I have no clue how to reassure the correct reference point. Klipper sends out or BedVisu just read the final interpolated mesh offsets value not the actual measured ones. I got a mesh 5x5 and in BedVisu I see data for 12x12. It would be great to only see the measured point and not the interpolated mess. Does anyone have an idea how to see only the measured values or how to find out if 0;0 is showing the correct values?

jneilliii, 2021-04-23:
currently there is no way in the plugin to have exact measured points if provided by the plugin either during the probing process, or from a report.

tarsonis123, 2021-04-23:
Hm, that's too bad. Since Klipper its wild guessing especially with glass and no use for BedVisu since then. The shown mesh position values in BedVisu don't collerate with the actual taken mesh coordinates. Its X0,X20,X59 and so on. And really measured the first point as X39. It would be great to have a feature like "move to origin mesh point" or "move to mesh 0;0". Scientific values are useless without any reference.

jneilliii, 2021-04-23:
The cursor arrow mover icon in the modebar of the graph can be clicked and then click on 0,0 point in the graph, but that's just going to send the nozzle to 0,0. That doesn't really help with identifying relative positioning. So I guess what Klipper is returning as a mesh is the interpolated mesh and not the actual probed point offsets. I assume that would be similar to the catmull subdivision meshes in old Marlin.

Curtis-Fletcher, 2021-05-01:
The lack of default-display rotation was driving me crazy as well.

As hacky as it is I ended up editing /static/js/bedlevelvisualizer.js manually on my octopi, setting scene.camera.eye (Line 235) to {x:0, y:2, z:1} gives the the expected rotation for my Ender 5, I just have to remember to redo that every time the plugin updates.

jneilliii, 2021-05-01:
@Curtis-Fletcher that moves the viewing angle, but doesn't change the axis labels correctly does it?

Curtis-Fletcher, 2021-05-01:
My axis labels are correct for my ender 5 it's just that 0,0 (Which is back, right on my ender 5) was positioned front, centre in rotation of the graph, what I posed above places 0,0 in the back right of the graph

There is already a setting for rotating the data WRT the axes if your kilpper setup is different. Then change X and Y (as above) Between  2/-2 to rotate the view. Between the two you should be able to handle any setup.

When I got confused I'd set a 1mm plate off centre on one side of the bed to give a hard identifier matching the physical to the display.

jneilliii, 2021-05-02:
Added a camera position feature in the latest Release Candidate version. This version also has remove numpy dependency and would love to get feedback on those changes as well to make sure it's not breaking something else. You can change the release channel in OctoPrint's Software Update settings for the plugin to `Release Candidate` and install the update when prompted. The field is in the format of `x,y,z` for the camera position, and testing with your values I get this.

[image]

[image]

Curtis-Fletcher, 2021-05-03:
I've switched to the RC channel and updated. The camera settings work wonderfully and I've had no issues with the plugin so far 
[image]

closed 2021-07-31 by jneilliii