jneilliii/OctoPrint-BedLevelVisualizer #382: [BUG]: Install downgraded numpy

sslupsky, 2021-02-17:
Installed this plugin today after installation of a few other plugins.  I noticed the following log messages:

```
Installing collected packages: numpy, Bed-Visualizer
Attempting uninstall: numpy
Found existing installation: numpy 1.20.1
Uninstalling numpy-1.20.1:
Successfully uninstalled numpy-1.20.1
Successfully installed Bed-Visualizer-1.0.0 numpy-1.19.2
Done!
```

Is appears this plugin downgraded numpy.  Is this intended?  I suggest this behaviour is unusual and may break something else that actually requires the more recent version of numpy.

jneilliii, 2021-02-17:
I had locked the numpy version because of a broken build on pypi wheels that was causing issues. It should work fine with pretty much any version of numpy, so if you want to upgrade it back, you could do that manually.

jneilliii, 2022-01-01:
no longer valid since removing numpy

closed 2022-01-01 by jneilliii