jneilliii/OctoPrint-BedLevelVisualizer #391: just updated octoprint to latest with python 3

turbodlr, 2021-02-24:
after updating octoprint to latest with python 3 I cannot use bed visualizer as when installed the octoprint reboots into safe mode.

turbodlr, 2021-02-24:
OctoPrint version : 1.5.3
OctoPi version    : 0.17.0

jneilliii, 2021-02-24:
Did you do this listed in the known issues section for python 3? 

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer#known-issues

It shouldn't effect OctoPrint from starting however so it may be a larger problem. Would have to see your octoprint.log to know for sure.

turbodlr, 2021-02-24:
Do you want me to install it again then after it restarts and goes into safe mode attach the log?

jneilliii, 2021-02-24:
After running the command to install the dependency and then install the plugin. OctoPrint will want to restart, and it should restart normally. If it doesn't, grab the octoprint.log from the logging section of settings and share here.

turbodlr, 2021-02-24:
ok I ran sudo apt install libatlas3-base first and its working now.  I am pretty new to this and this is on ender  5 plus so I put this gcode in from your wiki,.. look correct to you?  Also I set center as origin checkbox as the printer homes to center of bed.  And I am not sure about the comments you made about temperature auto reporting. Is this something I need to set?

G28      ; home all axes
M155 S30 ; reduce temperature reporting rate to reduce output pollution
@BEDLEVELVISUALIZER	; tell the plugin to watch for reported mesh
G29 T	   ; run bilinear probing
M155 S3  ; reset temperature reporting

jneilliii, 2021-02-24:
You don't need the center origin option and I've learned recently that bilinear leveling doesn't work with `G29 T` anymore so you may have to adjust your settings to something like the below. Try your settings first and if you get a weird message or it just doesn't graph try these other settings. The temperature reporting might not be a problem, but we won't know until we try.

```
G28
G29
@BEDLEVELVISUALIZER
M420 V
```

turbodlr, 2021-02-24:
Ok using what I originally posted works..  I think we are good.  Thanks for the replies!

closed 2021-02-24 by jneilliii