jneilliii/OctoPrint-BedLevelVisualizer #394: [BUG]:

joseaguardia, 2021-02-26:
**Describe the bug**
Every time I update the mesh, this plugin modifies the octoprint config.yml file, deleting everything related to the Custom Controls plugin.

**Expected behavior**
That the plugin does not modify config.yml

**Debug Log**
There is nothing in the log

**Firmware and Version**
Octoprint 1.5.3

jneilliii, 2021-02-26:
Install my patched Custom Controls plugin and that won't happen. The issue is with how that plugin loads/saves its settings. You can copy/paste the URL below into Plugin Manager > Get More > ...from URL over your existing install.

```
https://github.com/jneilliii/octoprint-customControl/archive/devel.zip
```

optiminimal, 2021-03-05:
Hi, I am still experiencing the same issue. 

I uninstalled and re-installed the patched customControl plugin (using the `devel` branch zip URL) your last commit in that branch shows in the plugin folder so it is definitely on the right version, but customControl `config.yml` are still being overwritten after a bed mesh save.

Tested on: 
OctoPrint 1.5.3 & Python 3.9.2

Can you do a double check on your end?

jneilliii, 2021-03-05:
I do have plans to remove that mesh save outside of config.yaml in a future release, but when I previously tested that devel branch it did resolve the problem. The only other thing I can think of is doing the following. Use Custom Control plugin to configure your controls the way you like them, save the settings, and then disable the plugin completely after saving. My plugin does not touch Custom Controls in config.yaml and if this is still happening it's entirely the fault of the other plugin and how it's loading/saving it's settings. Custom Controls is a native feature of OctoPrint, the plugin just gives you the ability to do it via the UI so your controls will remain.

optiminimal, 2021-03-07:
This is great! I am embarrassed to say that I did not know it was a native feature 😆  

Confirmed, removing the customControl plugin and managing it manually via config.yaml works. Thanks @jneilliii

closed 2021-03-07 by jneilliii