jneilliii/OctoPrint-BedLevelVisualizer #397: Bed Visualizer Error - "Overload resolution failed"

ItsDLite, 2021-02-27:
Bed Visualizer Error
Errors while attempting render of mesh data.
Error:
TypeError: Failed to execute 'uniformMatrix4fv' on 'WebGLRenderingContext': Overload resolution failed.
Received Data:
[object Object]

The printer is a LotMaxx Shark. I have no issues with my Ender. The GCode to start the auto bed leveling works fine. For me its G28 L 

I'd appreciate some advice!

jneilliii, 2021-02-27:
do you know what firmware is running on the lotmaxx? We need to figure out what the gcode command is for reporting the mesh offsets not just performing the probing.

ItsDLite, 2021-02-28:
Firmware is Configuration.txt 
>Para_Version	0.04	#configuration version eg: 0.01 0.02 1.02
If your asking Marlin version I'm not sure what the default is. 

Here is the GCODE used CURA they use in the LotMaxx machine profile if it helps any: 

G28 L;Home

G92 E0 ;Reset Extruder
G1 Z10.0 F3000 ;Move Z Axis up
;lotmaxx-dule-start
G1 E100 F600 ;E move 100
G92 E0 ;Reset Extruder
;lotmaxx-dule-end
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up

I appreciate the help!!

jneilliii, 2021-02-28:
Try this, manually send these commands from the terminal tab in octoprint and report back what each returns. 

```
G28 L
G29 T
M420 V
```

ItsDLite, 2021-02-28:
G28 L Returns:
Send: G28 L
Recv:  B@:0
Recv: 
Recv: -1.428750
Recv: -1.565000
Recv: -1.442500
Recv: -1.413750
Recv: -1.476250
Recv: -1.388750
Recv: -1.196250
Recv: -1.360000
Recv: -1.286250
Recv: -1.428750
Recv: -1.565000
Recv: -1.442500
Recv: -1.388750
Recv: -1.476250
Recv: -1.413750
Recv: -1.196250
Recv: -1.360000
Recv: -1.286250
Recv: ok

G29 T gives me:

Send: G29 T
Recv: ok

M420 V gives me:
Send: M420 V
Recv: ok

If I put all three into the "mesh update process" in the Octoprint settings I get: 
"Please wait, retrieving current mesh." and just keeps spinning around and then fails with:

TypeError: Failed to execute 'uniformMatrix4fv' on 'WebGLRenderingContext': Overload resolution failed.
Received Data:
[object Object]

jneilliii, 2021-03-01:
Yeah, the firmware isn't really reporting the probe points in a usable way for the plugin to interpret. If you can find any documentation on their supported gcode commands it would help.

ItsDLite, 2021-03-01:
No GCode instructions online or in the manual it came with. I requested it from the manufacture :( 
Figures crossed I ever get that

closed 2021-03-23 by github-actions[bot]