jneilliii/OctoPrint-BedLevelVisualizer #399: [BUG]: Missing parts of leveling graph

lightmaster, 2021-02-28:
**Describe the bug**
Bed mesh extends from -99 to 99 on both X and Y, yet the graph only extends from -78 to 100. Graph is also missing the corners entirely.

**Expected behavior**
Graph should display all the data that is shown in the topology map.

**Debug Log**
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/6058261/plugin_bedlevelvisualizer_debug.log)

Bed Mesh:
```gcode
G29 T

Bed Topography Report:

    (  -99, 99)                                                              ( 99, 99)
        0       1       2       3       4       5       6       7       8       9
 9 | +0.073   0.000   0.000  +0.025  +0.115  +0.170  +0.070  +0.034  +0.034  -0.100
   |
 8 | +0.060  +0.030  +0.015  +0.035  +0.016  +0.066  +0.040  +0.035  -0.100  -0.100
   |
 7 | +0.047  +0.055  +0.023  +0.037  +0.014  +0.040  +0.005  -0.004  -0.095  -0.100
   |
 6 | +0.260  +0.115  +0.069  +0.037  +0.021  +0.023  -0.001  -0.011  -0.090  -0.215
   |
 5 | +0.300  +0.179  +0.087  +0.048  +0.018  +0.007  -0.010  -0.034  -0.097  -0.220
   |
 4 | +0.365  +0.225  +0.125  +0.055 [+0.027] +0.011  +0.009  -0.031  -0.084  -0.225
   |
 3 | +0.390  +0.225  +0.144  +0.052  +0.039  +0.009  -0.001  -0.038  -0.075  -0.225
   |
 2 | +0.236  +0.195  +0.104  +0.028  +0.026  -0.001  +0.008  -0.025  -0.035  -0.035
   |
 1 | +0.330  +0.170  +0.020  -0.025  -0.013  -0.023  +0.010  +0.025  +0.005  +0.005
   |
 0 | +0.424  +0.045  +0.010  -0.025  -0.090  -0.050  +0.012  +0.070  +0.129  +0.045
        0       1       2       3       4       5       6       7       8       9
    (  -99,  -99)                                                              ( 99,  -99)

ok
```
**Screenshots**
[image]

**Firmware and Version**
Marlin bugfix-2.0.x branch, updated 2021-02-27
Bedvisualizer 1.0.1

jneilliii, 2021-03-01:
This might be the new circular stuff not interpreting properly?

lightmaster, 2021-03-01:
This bug has been around for a bit, just finally got around to writing up the bug report for it. The new circular changes didn't seem to effect it any.

jneilliii, 2021-03-01:
Interesting, the data isn't getting masked right for the circle, although @sofger's PR seems to work properly based on their comment [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/pull/393#issuecomment-786171971). What I am seeing in your debug log though is that the data might be getting corrupted every other row, which is causing the final mesh grid to not be the right dimensions.

```
[2021-02-28 18:20:03,289] DEBUG: mesh collection started
[2021-02-28 18:20:03,304] DEBUG: []
[2021-02-28 18:20:03,327] DEBUG: ['+0.073', '0.000', '0.000', '+0.025', '+0.115', '+0.170', '+0.070', '+0.034', '+0.034', '-0.100']
[2021-02-28 18:20:03,329] DEBUG: []
[2021-02-28 18:20:03,331] DEBUG: ['+0.060', '+0.030', '+0.015', '+0.035', '+0.016', '+0.066', '+0.040', '+0.035', '-0.100', '-0.100']
[2021-02-28 18:20:03,333] DEBUG: []
[2021-02-28 18:20:03,336] DEBUG: ['+0.047', '+0.055', '+0.023', '+0.037', '+0.014', '+0.040', '+0.005', '-0.004', '-0.095', '-0.100']
[2021-02-28 18:20:03,337] DEBUG: []
[2021-02-28 18:20:03,341] DEBUG: ['+0.260', '+0.115', '+0.069', '+0.037', '+0.021', '+0.023', '-0.001', '-0.011', '-0.090', '-0.215']
[2021-02-28 18:20:03,342] DEBUG: []
[2021-02-28 18:20:03,344] DEBUG: ['+0.300', '+0.179', '+0.087', '+0.048', '+0.018', '+0.007', '-0.010', '-0.034', '-0.097', '-0.220']
[2021-02-28 18:20:03,345] DEBUG: []
[2021-02-28 18:20:03,348] DEBUG: ['+0.365', '+0.225', '+0.125', '+0.055', '+0.027', '+0.011', '+0.009', '-0.031', '-0.084', '-0.225']
[2021-02-28 18:20:03,350] DEBUG: []
[2021-02-28 18:20:03,353] DEBUG: ['+0.390', '+0.225', '+0.144', '+0.052', '+0.039', '+0.009', '-0.001', '-0.038', '-0.075', '-0.225']
[2021-02-28 18:20:03,354] DEBUG: []
[2021-02-28 18:20:03,358] DEBUG: ['+0.236', '+0.195', '+0.104', '+0.028', '+0.026', '-0.001', '+0.008', '-0.025', '-0.035', '-0.035']
[2021-02-28 18:20:03,359] DEBUG: []
[2021-02-28 18:20:03,362] DEBUG: ['+0.330', '+0.170', '+0.020', '-0.025', '-0.013', '-0.023', '+0.010', '+0.025', '+0.005', '+0.005']
[2021-02-28 18:20:03,364] DEBUG: []
[2021-02-28 18:20:03,365] DEBUG: ['+0.424', '+0.045', '+0.010', '-0.025', '-0.090', '-0.050', '+0.012', '+0.070', '+0.129', '+0.045']
```

Could you share your firmware, version, and leveling type if you know as well as what's returned on the terminal tab when you run the mesh report command? What are you using for your settings for reporting the mesh?

lightmaster, 2021-03-01:
Marlin, bugfix-2.0.x compiled on 2021-02-27
Using UBL

I've been using the default of `G29 T` to report the topological map, however I'm now wondering how the plugin would handle using `G29 T1` which should send back a delimited map meant for being machine readable. Can't test that atm cuz I've got another 2.5 hours on my print, but will report back with the `G29 T1` output once that's done. The `G29 T` output is in my initial bug report up there.

lightmaster, 2021-03-01:
Also, just looked at the other PR you linked, and I'm not exactly sure why anyone would ever want to see a portion of their mesh and not the whole mesh. Even on Delta's the whole square mesh is used since it used the points outside of the circle to extrapolate the offset inbetween points along the edge of the circle. Could we at least get the circular cutout as a option that is disabled by default and can be enabled by the user?

jneilliii, 2021-03-01:
The circle was supposed to be there from earlier versions, but in those cases the firmware didn't report those extrapolated points on a rectangular grid like the do on newer versions. When I first programmed this plugin, it was designed around G29 T1 with UBL, because I believe that's a csv format. Either way, I think if you change your settings to use M420 V just after the @BEDLEVELVISUALIZER command it will write it out a single stream that the plugin should be able to interpret as well.

lightmaster, 2021-03-01:
Yea... Looking at the example codes again, I'm pretty sure this is a PEBKAC issue. I was using Bilinear Leveling and switched to UBL, and never changed the code to grab the mesh data. 1 more hour on the print and then I'll let you know.

Maybe for future idiots like me, a way to detect incorrect settings for getting the mesh, such as getting the human-readable map instead of the delimited map, and throw up a warning?

[image]

lightmaster, 2021-03-01:
All ways that I can output the mesh data cause this issue and produce the same graph that's in the original message up there. Here's the various output formats

```gcode
G29 T1

Bed Topography Report for CSV:

0.085	0.010	0.008	0.032	-0.005	0.048	0.071	0.034	0.032	-0.104
0.071	0.040	-0.052	-0.024	0.021	0.029	0.041	-0.041	-0.102	-0.104
0.057	-0.011	-0.019	0.042	-0.008	0.043	0.031	-0.030	-0.148	-0.105
0.270	0.048	0.051	0.042	0.034	0.050	0.014	-0.028	-0.144	-0.220
0.260	0.127	0.063	0.067	0.031	0.023	0.014	-0.036	-0.161	-0.361
0.374	0.182	0.101	0.069	0.029	0.011	-0.007	-0.049	-0.164	-0.356
0.399	0.212	0.109	0.056	0.020	0.009	-0.002	-0.082	-0.180	-0.232
0.244	0.226	0.083	0.006	-0.008	-0.027	-0.034	-0.094	-0.161	-0.043
0.337	0.176	0.049	-0.048	-0.087	-0.065	-0.043	-0.005	-0.001	-0.003
0.431	0.050	0.013	-0.023	-0.155	-0.087	0.009	0.065	0.122	0.036
ok
```

```gcode
M420 T0 V

Bed Topography Report:

    (  -99, 99)                                                              ( 99, 99)
        0       1       2       3       4       5       6       7       8       9
 9 | +0.085  +0.010  +0.008  +0.032  -0.005  +0.048  +0.071  +0.034  +0.032  -0.104
   |
 8 | +0.071  +0.040  -0.052  -0.024  +0.021  +0.029  +0.041 [-0.041] -0.102  -0.104
   |
 7 | +0.057  -0.011  -0.019  +0.042  -0.008  +0.043  +0.031  -0.030  -0.148  -0.105
   |
 6 | +0.270  +0.048  +0.051  +0.042  +0.034  +0.050  +0.014  -0.028  -0.144  -0.220
   |
 5 | +0.260  +0.127  +0.063  +0.067  +0.031  +0.023  +0.014  -0.036  -0.161  -0.361
   |
 4 | +0.374  +0.182  +0.101  +0.069  +0.029  +0.011  -0.007  -0.049  -0.164  -0.356
   |
 3 | +0.399  +0.212  +0.109  +0.056  +0.020  +0.009  -0.002  -0.082  -0.180  -0.232
   |
 2 | +0.244  +0.226  +0.083  +0.006  -0.008  -0.027  -0.034  -0.094  -0.161  -0.043
   |
 1 | +0.337  +0.176  +0.049  -0.048  -0.087  -0.065  -0.043  -0.005  -0.001  -0.003
   |
 0 | +0.431  +0.050  +0.013  -0.023  -0.155  -0.087  +0.009  +0.065  +0.122  +0.036
        0       1       2       3       4       5       6       7       8       9
    (  -99,  -99)                                                              ( 99,  -99)

Mesh is valid
Storage slot: 0
echo:Bed Leveling ON
echo:Fade Height 10.00
ok
```

```gcode
M420 T1 V

Bed Topography Report for CSV:

0.085	0.010	0.008	0.032	-0.005	0.048	0.071	0.034	0.032	-0.104
0.071	0.040	-0.052	-0.024	0.021	0.029	0.041	-0.041	-0.102	-0.104
0.057	-0.011	-0.019	0.042	-0.008	0.043	0.031	-0.030	-0.148	-0.105
0.270	0.048	0.051	0.042	0.034	0.050	0.014	-0.028	-0.144	-0.220
0.260	0.127	0.063	0.067	0.031	0.023	0.014	-0.036	-0.161	-0.361
0.374	0.182	0.101	0.069	0.029	0.011	-0.007	-0.049	-0.164	-0.356
0.399	0.212	0.109	0.056	0.020	0.009	-0.002	-0.082	-0.180	-0.232
0.244	0.226	0.083	0.006	-0.008	-0.027	-0.034	-0.094	-0.161	-0.043
0.337	0.176	0.049	-0.048	-0.087	-0.065	-0.043	-0.005	-0.001	-0.003
0.431	0.050	0.013	-0.023	-0.155	-0.087	0.009	0.065	0.122	0.036
Mesh is valid
Storage slot: 0
echo:Bed Leveling ON
echo:Fade Height 10.00
ok
```

```gcode
M420 T4 V

Bed Topography Report:

    (  -99, 99)                                                              ( 99, 99)
        0       1       2       3       4       5       6       7       8       9
 9 | +0.085  +0.010  +0.008  +0.032  -0.005  +0.048  +0.071  +0.034  +0.032  -0.104
 8 | +0.071  +0.040  -0.052  -0.024  +0.021  +0.029  +0.041 [-0.041] -0.102  -0.104
 7 | +0.057  -0.011  -0.019  +0.042  -0.008  +0.043  +0.031  -0.030  -0.148  -0.105
 6 | +0.270  +0.048  +0.051  +0.042  +0.034  +0.050  +0.014  -0.028  -0.144  -0.220
 5 | +0.260  +0.127  +0.063  +0.067  +0.031  +0.023  +0.014  -0.036  -0.161  -0.361
 4 | +0.374  +0.182  +0.101  +0.069  +0.029  +0.011  -0.007  -0.049  -0.164  -0.356
 3 | +0.399  +0.212  +0.109  +0.056  +0.020  +0.009  -0.002  -0.082  -0.180  -0.232
 2 | +0.244  +0.226  +0.083  +0.006  -0.008  -0.027  -0.034  -0.094  -0.161  -0.043
 1 | +0.337  +0.176  +0.049  -0.048  -0.087  -0.065  -0.043  -0.005  -0.001  -0.003
 0 | +0.431  +0.050  +0.013  -0.023  -0.155  -0.087  +0.009  +0.065  +0.122  +0.036
        0       1       2       3       4       5       6       7       8       9
    (  -99,  -99)                                                              ( 99,  -99)

Mesh is valid
Storage slot: 0
echo:Bed Leveling ON
echo:Fade Height 10.00
ok
```