jneilliii/OctoPrint-BedLevelVisualizer #405: [FR] API for OctoDash

hawkeyexp, 2021-03-11:
Hi, i was asking for support of G29 (showing info while leveling is running and possible the results) and the dev told me it is hard for him cause octoprint is not aware of the current code.
Your plugin does this trick :-) Would it be possible to release a simple access for octodash to catch the informations like start of leveling end of leveling and results - possible also the heatmap ?

Here is the irginal issue for octodash:

https://github.com/UnchartedBull/OctoDash/issues/1531

Thx

jneilliii, 2021-03-11:
Interestingly enough I think my [BLVLogger](https://github.com/jneilliii/OctoPrint-BLVLogger) plugin might work in this situation if @UnchartedBull just needs an API endpoint to query. It's a little convoluted but it assumes a start and end date to pull a list of historical meshes.

I know that he is already in the process of converting over to websockets for communicating with OctoPrint and when that happens he can actually listen in on the data that my plugin is sending back to the front end natively [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/a8422a77215a6b5f01eac95186c8be2d7d1d069d/octoprint_bedlevelvisualizer/__init__.py#L410) without this additional step.

UnchartedBull, 2021-03-11:
That is great news! The websocket transition is going really well so far. Only Job Progress and Z-Height left 🤞 . I guess this issue can be closed then and I would come back to the original OctoDash issue once v3 is released. 

Thanks for all your work on the various plugins! :)

jneilliii, 2021-03-12:
Oh, and BTW a lot of plugin utilize that send plugin message method, so you might be able to integrate a lot of plugin related information once that transition to websocket is complete (ie Tasmota on/off status).

UnchartedBull, 2021-03-12:
The PSU Extension should be even simpler once https://github.com/UnchartedBull/OctoDash/pull/1438 is merged. Only requires a single function without an additional service from now on. https://github.com/UnchartedBull/OctoDash/blob/feature/websocket/src/app/services/enclosure/enclosure.octoprint.service.ts#L77 those are the releavant lines :)

jneilliii, 2021-03-12:
Interesting, you're putting all those inside an enclosure service. That does look like it will be easier to implement. I'll base my Tasmota changes on that branch if that's where you are headed.