jneilliii/OctoPrint-BedLevelVisualizer #407: [FR] Get current mesh from printer

hawkeyexp, 2021-03-17:
Hi, is it possible to add an option to get current mesh from printer which was possible triggered to create manually or via another control unit like octodash?

I know G29 S0 will report curent mesh - for me it would make sense to catch curent mesh if plugin is opened or an aditional button to request current mesh from printer.

Would be nice.

jneilliii, 2021-03-17:
You can use M420 V in some cases, but the trick is that you have to precede it with the `@BEDLEVELVISUALIZER` flag or by preceding the command that reports a mesh using the command `M117 E1 BEDLEVELVISUALIZER` to basically tell the plugin to start recording what the printer is saying.

jneilliii, 2021-03-17:
oh, and your printer has to support the sending of echo commands from the printer, that's what the `M118 E1 BEDLEVELVISUALIZER` command does.

https://marlinfw.org/docs/gcode/M118.html

hawkeyexp, 2021-03-17:
That's celar how it works :-)
But it would be nice for example to have two buttons with custom g-code (one for auto leveling with recording the mesh) and one to only request the current mesh or an auto loading of current mesh while opening the tab in gui :-)

jneilliii, 2021-03-17:
This is possible too with custom controls. You set the default one for the update mesh button to use the command for reporting the mesh only and then add a custom button in the plugin's settings for running the mesh process completely and then reporting.

closed 2021-04-08 by github-actions[bot]