jneilliii/OctoPrint-BedLevelVisualizer #414: [BUG]: BLV deletes my custom controls

wlbillmartin, 2021-03-26:
When I run the BLV, my custom controls disappear.
When looking at config.yaml, they're gone.
Luckly, I have a copy of my controls and can re-paste them into the config.yaml, so they're not gone forever, but this is a pain to have to paste them in whenever I run BLV.
When Bed Leveling is run from BLTouch plugin, all is okay. Same when bed leveling (BLTouch) (G29) is run before a print.
This only happens with BLV is run.

Otherwise, I LOVE BLV!!!

jneilliii, 2021-03-26:
That's been discussed before, and documented in the known issues [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer#known-issues). It's actually a bug in the Custom Control plugin. I have a PR that hasn't been merged that will resolve this. You can install it by copying/pasting the URL below in the Plugin Manager > Get More > ...from URL and clicking Install. If the plugin author over there ever updates their code you'll still get an upgrade notice. BTW you can also just disable that plugin once you get your custom control the way you like them.

```
https://github.com/jneilliii/octoprint-customControl/archive/refs/heads/devel.zip
```

closed 2021-03-26 by jneilliii

wlbillmartin, 2021-03-26:
Well… I am an idiot! One of the first things we learn in Aviation is: If you do something and then things go awry, undo what you last did! (this would have saved everybody in the 737 Max accidents)

I disabled that plugin and all is good. I even tested re-enabling it and making a change and then disabling it and that worked as well.

So I’m happy, I can make custom controls when I need to and all is good.

THANK YOU!!!

And please accept my apologies for not seeing this in the “Known Issues” listing.

< sigh >
 
Sir William Louis "Bill" Martin Jr., ATP, Capt. (Ret.), OMS
Active Participant - Planet Earth
AKA:
William P. Lewis
Comedian/Radio Personality/Actor
http://www.williamplewis.com/
Phone: 949.933.6230
Fax: 909.510.8253

 

https://github.com/jneilliii/octoprint-customControl/archive/refs/heads/devel.zip