jneilliii/OctoPrint-BedLevelVisualizer #420: [BUG]: Topology Graph is Missing

fractalsauce, 2021-04-15:
**Describe the bug**
All features of the plugin functions as expected except that the topology graph is completely missing.

This seems to have happened out of nowhere. A week ago everything was working perfectly. I left for a week's vacation and of course turned my printer off while I was gone. When I came back and booted the printer and OctoPrint back up, the topology graph was simply no longer there. I am able to click the Update Mesh Now button and the actual mesh is indeed updated, it's just the graph doesn't display. 

I have tried rebooting the Raspberry Pi from within the OctoPrint page as well as pressing CTRL+F5 to clear the browser cache when reloading the window to no success.

**Expected behavior**
To be able to view a 3d topology graph of the probed bed mesh.

**Debug Log**
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/6320630/plugin_bedlevelvisualizer_debug.log)

**Screenshots**
[image]
[image]

**Firmware and Version**
Marlin 2.0.7.2

jneilliii, 2021-04-15:
Does this also happen in incognito mode of your browser?

fractalsauce, 2021-04-15:
Hmm out of curiosity I just tried Google Chrome and it works fine.

I guess maybe Firefox or one of my add-ons had an update that broke it.

Well that sucks, I use Firefox for everything and only use Chrome for testing websites that I have issues with in Firefox and for the auto-translate function for websites that have no English version.

I'm not sure if this is something that you could fix with an update of the plugin or not. If you could that'd be awesome.

Anyways I guess technically my issue is fixed.

jneilliii, 2021-04-15:
so what is required is webgl/opengl for the surface graph to be rendered. Maybe there is a setting that you can change that will allow that type of operation on the site.

fractalsauce, 2021-04-16:
So I have no idea what the issue was, but after I rebooted my computer, the graph is back. I feel a little silly that I didn't try that first. I also went ahead and followed [this guide](https://www.interplaylearning.com/help/how-to-enable-webgl-in-firefox) to ensure that webgl is enabled in Firefox and it is.

Going to go ahead and close the issue. Thanks for your help!

closed 2021-04-16 by fractalsauce