jneilliii/OctoPrint-BedLevelVisualizer #424: [BUG]: The latest update hangs after pressing "update mesh now" now

tigercjn, 2021-05-06:
**Describe the bug**

**Expected behavior**

**Debug Log**

**Screenshots**

**Firmware and Version**

jneilliii, 2021-05-06:
do you mean 1.1.0rc1 was working fine for you and the rc2 broke it, or switching from 1.0.1 to the rc version did it?

tigercjn, 2021-05-06:
I was on the 1.0.1 version then switched as octoprint said there was an update.

jneilliii, 2021-05-06:
If you go into OctoPrint's software update settings is the plugin's release channel configured for Release Candidate or Stable?

jneilliii, 2021-05-06:
Please also share your full debug log as an attachment so I can get a full picture of what's happening. You can just drag it into a comment here.

tigercjn, 2021-05-07:
It was set to release the candidate.

Cheers Chris.

tigercjn, 2021-05-07:
Full Log.
[2021-05-06 12:25:32,193] DEBUG: mesh collection started
[2021-05-06 12:25:32,253] DEBUG: resetting mesh to blank because of CATMULL subdivision
[2021-05-06 12:25:32,270] DEBUG: ['+6.21000', '+6.23366', '+6.25731', '+6.28750', '+6.33528', '+6.38958', '+6.42375', '+6.42347', '+6.40305', '+6.37875', '+6.35468', '+6.32671', '+6.29875']
[2021-05-06 12:25:32,273] DEBUG: ['+6.40250', '+6.43894', '+6.47539', '+6.51088', '+6.54996', '+6.58808', '+6.61069', '+6.61174', '+6.59727', '+6.57083', '+6.52963', '+6.47646', '+6.42329']
[2021-05-06 12:25:32,280] DEBUG: ['+6.59500', '+6.64423', '+6.69346', '+6.73426', '+6.76464', '+6.78659', '+6.79764', '+6.80001', '+6.79148', '+6.76291', '+6.70458', '+6.62620', '+6.54782']
[2021-05-06 12:25:32,287] DEBUG: ['+6.74000', '+6.79620', '+6.85241', '+6.89500', '+6.91699', '+6.92537', '+6.92750', '+6.93227', '+6.93078', '+6.90375', '+6.83620', '+6.74310', '+6.65000']
[2021-05-06 12:25:32,298] DEBUG: ['+6.81379', '+6.86584', '+6.91788', '+6.95495', '+6.96725', '+6.96458', '+6.96134', '+6.97118', '+6.98045', '+6.96268', '+6.90002', '+6.81031', '+6.72060']
[2021-05-06 12:25:32,310] DEBUG: ['+6.84009', '+6.88216', '+6.92423', '+6.95227', '+6.95518', '+6.94406', '+6.93810', '+6.95408', '+6.97521', '+6.97037', '+6.92051', '+6.84467', '+6.76884']
[2021-05-06 12:25:32,320] DEBUG: ['+6.84250', '+6.87893', '+6.91537', '+6.93875', '+6.93773', '+6.92366', '+6.91750', '+6.93694', '+6.96430', '+6.96750', '+6.92778', '+6.86389', '+6.80000']
[2021-05-06 12:25:32,329] DEBUG: ['+6.82810', '+6.86771', '+6.90731', '+6.93435', '+6.93866', '+6.93040', '+6.92750', '+6.94500', '+6.96785', '+6.96889', '+6.93176', '+6.87280', '+6.81384']
[2021-05-06 12:25:32,340] DEBUG: ['+6.78981', '+6.83693', '+6.88404', '+6.91912', '+6.93420', '+6.93725', '+6.94014', '+6.95302', '+6.96574', '+6.95972', '+6.92251', '+6.86655', '+6.81060']
[2021-05-06 12:25:32,352] DEBUG: ['+6.73000', '+6.78569', '+6.84139', '+6.88500', '+6.91000', '+6.92291', '+6.93125', '+6.94134', '+6.94685', '+6.93625', '+6.90032', '+6.84828', '+6.79625']
[2021-05-06 12:25:32,362] DEBUG: ['+6.64227', '+6.70689', '+6.77151', '+6.82301', '+6.85469', '+6.87324', '+6.88560', '+6.89684', '+6.90188', '+6.89245', '+6.86097', '+6.81502', '+6.76907']
[2021-05-06 12:25:32,371] DEBUG: ['+6.53301', '+6.60763', '+6.68226', '+6.74213', '+6.77963', '+6.80238', '+6.81842', '+6.83263', '+6.84013', '+6.83435', '+6.80868', '+6.76973', '+6.73078']
[2021-05-06 12:25:32,382] DEBUG: ['+6.42375', '+6.50838', '+6.59301', '+6.66125', '+6.70458', '+6.73153', '+6.75125', '+6.76842', '+6.77838', '+6.77625', '+6.75639', '+6.72444', '+6.69250']
[2021-05-06 12:25:32,389] DEBUG: {'type': 'rectangular', 'x_min': -10.0, 'x_max': 320.0, 'y_min': -10.0, 'y_max': 350.0, 'z_min': 0.0, 'z_max': 330.0}
[2021-05-06 12:25:32,389] DEBUG: stopping mesh collection
[2021-05-06 12:25:32,390] DEBUG: using relative offsets
[2021-05-06 12:27:30,205] DEBUG: Canceling mesh collection per user request
[2021-05-06 12:27:30,206] DEBUG: Mesh data collected prior to cancel:
[2021-05-06 12:27:30,206] DEBUG: [['+6.21000', '+6.23366', '+6.25731', '+6.28750', '+6.33528', '+6.38958', '+6.42375', '+6.42347', '+6.40305', '+6.37875', '+6.35468', '+6.32671', '+6.29875'], ['+6.40250', '+6.43894', '+6.47539', '+6.51088', '+6.54996', '+6.58808', '+6.61069', '+6.61174', '+6.59727', '+6.57083', '+6.52963', '+6.47646', '+6.42329'], ['+6.59500', '+6.64423', '+6.69346', '+6.73426', '+6.76464', '+6.78659', '+6.79764', '+6.80001', '+6.79148', '+6.76291', '+6.70458', '+6.62620', '+6.54782'], ['+6.74000', '+6.79620', '+6.85241', '+6.89500', '+6.91699', '+6.92537', '+6.92750', '+6.93227', '+6.93078', '+6.90375', '+6.83620', '+6.74310', '+6.65000'], ['+6.81379', '+6.86584', '+6.91788', '+6.95495', '+6.96725', '+6.96458', '+6.96134', '+6.97118', '+6.98045', '+6.96268', '+6.90002', '+6.81031', '+6.72060'], ['+6.84009', '+6.88216', '+6.92423', '+6.95227', '+6.95518', '+6.94406', '+6.93810', '+6.95408', '+6.97521', '+6.97037', '+6.92051', '+6.84467', '+6.76884'], ['+6.84250', '+6.87893', '+6.91537', '+6.93875', '+6.93773', '+6.92366', '+6.91750', '+6.93694', '+6.96430', '+6.96750', '+6.92778', '+6.86389', '+6.80000'], ['+6.82810', '+6.86771', '+6.90731', '+6.93435', '+6.93866', '+6.93040', '+6.92750', '+6.94500', '+6.96785', '+6.96889', '+6.93176', '+6.87280', '+6.81384'], ['+6.78981', '+6.83693', '+6.88404', '+6.91912', '+6.93420', '+6.93725', '+6.94014', '+6.95302', '+6.96574', '+6.95972', '+6.92251', '+6.86655', '+6.81060'], ['+6.73000', '+6.78569', '+6.84139', '+6.88500', '+6.91000', '+6.92291', '+6.93125', '+6.94134', '+6.94685', '+6.93625', '+6.90032', '+6.84828', '+6.79625'], ['+6.64227', '+6.70689', '+6.77151', '+6.82301', '+6.85469', '+6.87324', '+6.88560', '+6.89684', '+6.90188', '+6.89245', '+6.86097', '+6.81502', '+6.76907'], ['+6.53301', '+6.60763', '+6.68226', '+6.74213', '+6.77963', '+6.80238', '+6.81842', '+6.83263', '+6.84013', '+6.83435', '+6.80868', '+6.76973', '+6.73078'], ['+6.42375', '+6.50838', '+6.59301', '+6.66125', '+6.70458', '+6.73153', '+6.75125', '+6.76842', '+6.77838', '+6.77625', '+6.75639', '+6.72444', '+6.69250']]
[2021-05-06 12:27:30,207] DEBUG: Mesh data after clearing:
[2021-05-06 12:27:30,208] DEBUG: []
[2021-05-06 12:27:51,513] DEBUG: mesh collection started
[2021-05-06 12:27:51,532] DEBUG: ['+6.210', '+6.287', '+6.424', '+6.379', '+6.299']
[2021-05-06 12:27:51,536] DEBUG: ['+6.740', '+6.895', '+6.927', '+6.904', '+6.650']
[... cut ...]
[2021-05-07 11:40:35,371] DEBUG: ['+6.53301', '+6.60763', '+6.68226', '+6.74213', '+6.77963', '+6.80238', '+6.81842', '+6.83263', '+6.84013', '+6.83435', '+6.80868', '+6.76973', '+6.73078']
[2021-05-07 11:40:35,383] DEBUG: ['+6.42375', '+6.50838', '+6.59301', '+6.66125', '+6.70458', '+6.73153', '+6.75125', '+6.76842', '+6.77838', '+6.77625', '+6.75639', '+6.72444', '+6.69250']
[2021-05-07 11:40:35,389] DEBUG: {'type': 'rectangular', 'x_min': -10.0, 'x_max': 320.0, 'y_min': -10.0, 'y_max': 350.0, 'z_min': 0.0, 'z_max': 330.0}
[2021-05-07 11:40:35,389] DEBUG: stopping mesh collection
[2021-05-07 11:40:35,390] DEBUG: using relative offsets

tigercjn, 2021-05-07:
Moving back to the stable release works fine.

[2021-05-06 12:25:32,193] DEBUG: mesh collection started
[2021-05-06 12:25:32,253] DEBUG: resetting mesh to blank because of CATMULL subdivision
[2021-05-06 12:25:32,270] DEBUG: ['+6.21000', '+6.23366', '+6.25731', '+6.28750', '+6.33528', '+6.38958', '+6.42375', '+6.42347', '+6.40305', '+6.37875', '+6.35468', '+6.32671', '+6.29875']
[2021-05-06 12:25:32,273] DEBUG: ['+6.40250', '+6.43894', '+6.47539', '+6.51088', '+6.54996', '+6.58808', '+6.61069', '+6.61174', '+6.59727', '+6.57083', '+6.52963', '+6.47646', '+6.42329']
[2021-05-06 12:25:32,280] DEBUG: ['+6.59500', '+6.64423', '+6.69346', '+6.73426', '+6.76464', '+6.78659', '+6.79764', '+6.80001', '+6.79148', '+6.76291', '+6.70458', '+6.62620', '+6.54782']
[2021-05-06 12:25:32,287] DEBUG: ['+6.74000', '+6.79620', '+6.85241', '+6.89500', '+6.91699', '+6.92537', '+6.92750', '+6.93227', '+6.93078', '+6.90375', '+6.83620', '+6.74310', '+6.65000']
[2021-05-06 12:25:32,298] DEBUG: ['+6.81379', '+6.86584', '+6.91788', '+6.95495', '+6.96725', '+6.96458', '+6.96134', '+6.97118', '+6.98045', '+6.96268', '+6.90002', '+6.81031', '+6.72060']
[2021-05-06 12:25:32,310] DEBUG: ['+6.84009', '+6.88216', '+6.92423', '+6.95227', '+6.95518', '+6.94406', '+6.93810', '+6.95408', '+6.97521', '+6.97037', '+6.92051', '+6.84467', '+6.76884']
[2021-05-06 12:25:32,320] DEBUG: ['+6.84250', '+6.87893', '+6.91537', '+6.93875', '+6.93773', '+6.92366', '+6.91750', '+6.93694', '+6.96430', '+6.96750', '+6.92778', '+6.86389', '+6.80000']
[2021-05-06 12:25:32,329] DEBUG: ['+6.82810', '+6.86771', '+6.90731', '+6.93435', '+6.93866', '+6.93040', '+6.92750', '+6.94500', '+6.96785', '+6.96889', '+6.93176', '+6.87280', '+6.81384']
[2021-05-06 12:25:32,340] DEBUG: ['+6.78981', '+6.83693', '+6.88404', '+6.91912', '+6.93420', '+6.93725', '+6.94014', '+6.95302', '+6.96574', '+6.95972', '+6.92251', '+6.86655', '+6.81060']
[2021-05-06 12:25:32,352] DEBUG: ['+6.73000', '+6.78569', '+6.84139', '+6.88500', '+6.91000', '+6.92291', '+6.93125', '+6.94134', '+6.94685', '+6.93625', '+6.90032', '+6.84828', '+6.79625']
[2021-05-06 12:25:32,362] DEBUG: ['+6.64227', '+6.70689', '+6.77151', '+6.82301', '+6.85469', '+6.87324', '+6.88560', '+6.89684', '+6.90188', '+6.89245', '+6.86097', '+6.81502', '+6.76907']
[2021-05-06 12:25:32,371] DEBUG: ['+6.53301', '+6.60763', '+6.68226', '+6.74213', '+6.77963', '+6.80238', '+6.81842', '+6.83263', '+6.84013', '+6.83435', '+6.80868', '+6.76973', '+6.73078']
[2021-05-06 12:25:32,382] DEBUG: ['+6.42375', '+6.50838', '+6.59301', '+6.66125', '+6.70458', '+6.73153', '+6.75125', '+6.76842', '+6.77838', '+6.77625', '+6.75639', '+6.72444', '+6.69250']
[2021-05-06 12:25:32,389] DEBUG: {'type': 'rectangular', 'x_min': -10.0, 'x_max': 320.0, 'y_min': -10.0, 'y_max': 350.0, 'z_min': 0.0, 'z_max': 330.0}
[2021-05-06 12:25:32,389] DEBUG: stopping mesh collection
[2021-05-06 12:25:32,390] DEBUG: using relative offsets
[2021-05-06 12:27:30,205] DEBUG: Canceling mesh collection per user request
[2021-05-06 12:27:30,206] DEBUG: Mesh data collected prior to cancel:
[2021-05-06 12:27:30,206] DEBUG: [['+6.21000', '+6.23366', '+6.25731', '+6.28750', '+6.33528', '+6.38958', '+6.42375', '+6.42347', '+6.40305', '+6.37875', '+6.35468', '+6.32671', '+6.29875'], ['+6.40250', '+6.43894', '+6.47539', '+6.51088', '+6.54996', '+6.58808', '+6.61069', '+6.61174', '+6.59727', '+6.57083', '+6.52963', '+6.47646', '+6.42329'], ['+6.59500', '+6.64423', '+6.69346', '+6.73426', '+6.76464', '+6.78659', '+6.79764', '+6.80001', '+6.79148', '+6.76291', '+6.70458', '+6.62620', '+6.54782'], ['+6.74000', '+6.79620', '+6.85241', '+6.89500', '+6.91699', '+6.92537', '+6.92750', '+6.93227', '+6.93078', '+6.90375', '+6.83620', '+6.74310', '+6.65000'], ['+6.81379', '+6.86584', '+6.91788', '+6.95495', '+6.96725', '+6.96458', '+6.96134', '+6.97118', '+6.98045', '+6.96268', '+6.90002', '+6.81031', '+6.72060'], ['+6.84009', '+6.88216', '+6.92423', '+6.95227', '+6.95518', '+6.94406', '+6.93810', '+6.95408', '+6.97521', '+6.97037', '+6.92051', '+6.84467', '+6.76884'], ['+6.84250', '+6.87893', '+6.91537', '+6.93875', '+6.93773', '+6.92366', '+6.91750', '+6.93694', '+6.96430', '+6.96750', '+6.92778', '+6.86389', '+6.80000'], ['+6.82810', '+6.86771', '+6.90731', '+6.93435', '+6.93866', '+6.93040', '+6.92750', '+6.94500', '+6.96785', '+6.96889', '+6.93176', '+6.87280', '+6.81384'], ['+6.78981', '+6.83693', '+6.88404', '+6.91912', '+6.93420', '+6.93725', '+6.94014', '+6.95302', '+6.96574', '+6.95972', '+6.92251', '+6.86655', '+6.81060'], ['+6.73000', '+6.78569', '+6.84139', '+6.88500', '+6.91000', '+6.92291', '+6.93125', '+6.94134', '+6.94685', '+6.93625', '+6.90032', '+6.84828', '+6.79625'], ['+6.64227', '+6.70689', '+6.77151', '+6.82301', '+6.85469', '+6.87324', '+6.88560', '+6.89684', '+6.90188', '+6.89245', '+6.86097', '+6.81502', '+6.76907'], ['+6.53301', '+6.60763', '+6.68226', '+6.74213', '+6.77963', '+6.80238', '+6.81842', '+6.83263', '+6.84013', '+6.83435', '+6.80868', '+6.76973', '+6.73078'], ['+6.42375', '+6.50838', '+6.59301', '+6.66125', '+6.70458', '+6.73153', '+6.75125', '+6.76842', '+6.77838', '+6.77625', '+6.75639', '+6.72444', '+6.69250']]
[2021-05-06 12:27:30,207] DEBUG: Mesh data after clearing:
[2021-05-06 12:27:30,208] DEBUG: []
[2021-05-06 12:27:51,513] DEBUG: mesh collection started
[2021-05-06 12:27:51,532] DEBUG: ['+6.210', '+6.287', '+6.424', '+6.379', '+6.299']
[2021-05-06 12:27:51,536] DEBUG: ['+6.740', '+6.895', '+6.927', '+6.904', '+6.650']
[... cut ...]
[2021-05-07 11:40:35,371] DEBUG: ['+6.53301', '+6.60763', '+6.68226', '+6.74213', '+6.77963', '+6.80238', '+6.81842', '+6.83263', '+6.84013', '+6.83435', '+6.80868', '+6.76973', '+6.73078']
[2021-05-07 11:40:35,383] DEBUG: ['+6.42375', '+6.50838', '+6.59301', '+6.66125', '+6.70458', '+6.73153', '+6.75125', '+6.76842', '+6.77838', '+6.77625', '+6.75639', '+6.72444', '+6.69250']
[2021-05-07 11:40:35,389] DEBUG: {'type': 'rectangular', 'x_min': -10.0, 'x_max': 320.0, 'y_min': -10.0, 'y_max': 350.0, 'z_min': 0.0, 'z_max': 330.0}
[2021-05-07 11:40:35,389] DEBUG: stopping mesh collection
[2021-05-07 11:40:35,390] DEBUG: using relative offsets

jneilliii, 2021-05-07:
Yeah, I think something must be off with the relative offsets section. That version in release candidate removed numpy, which was using a utility function to do that for me. There must be something wrong with the replacement function. Those changes were contributed by someone else, so I'll take a look when I can to try and resolve. I know they were making additional updates to resolve some circular bed related logic as well after removing numpy.

Thanks for the logs, that will help reproduce your exact issue.

tablatronix, 2021-05-08:
I also updated and now it runs and doesn't actually update the graph or timestamp etc. ( not sure if that is what it means by hangs) level works, just does nothing when its done, just sits there. I will try to check that commit

tablatronix, 2021-05-08:
hmm how can I force it to pickup rc3?

jneilliii, 2021-05-08:
Cool  I think the issue was switching to a minimal plotly library, but forgetting to include it in the push and release.

tablatronix, 2021-05-15:
Still Failing