jneilliii/OctoPrint-BedLevelVisualizer #429: [BUG]: Adjustment Angles

cjmalchow, 2021-05-12:
Hi,

I'm not sure if this is a bug or not, but if I click on any of the adjustment angles in the chart the whole chart changes. I tried searching everywhere online for an answer as to what is happening but couldn't find anything. Clarification on this if it isn't a bug would be greatly appreciated. Thanks!

jneilliii, 2021-05-12:
The way in which the adjustment angle table was programmed by another contributor that is by design. It zeros out based on the point that you click in the table, making all the adjustments relative to that one point.

closed 2021-05-12 by jneilliii

cjmalchow, 2021-05-13:
So, hopefully I can clarify... let's say no adjustments could be made to one of the corners, you can click on it and it recalculates how you would have to adjust the others to get level?

cjmalchow, 2021-05-13:
My other question is how do you revert the table back to what it was at when it originally updated the mesh? Thanks so much, love the plugin and really appreciate the work you and the others do!

jneilliii, 2021-05-13:
> So, hopefully I can clarify... let's say no adjustments could be made to one of the corners, you can click on it and it recalculates how you would have to adjust the others to get level?

Correct, I believe that was thre intent. 

> My other question is how do you revert the table back to what it was at when it originally updated the mesh?

I honestly don't know. The angle adjustment section was contributed by another user and I don't personally use it.

cjmalchow, 2021-05-14:
Thanks for the reply, I appreciate it. I'm curious if anyone is working on
documentation right now for the plugin? If not, I wouldn't mind
volunteering to help out with it. I've never written documentation before,
but I think I could manage.

Let me know and thanks for what you do!

Cameron

jneilliii, 2021-05-14:
I'm completely open to any form of documentation that you would be willing to add. All of it currently is in this repo in the wiki sub-folder. It's done in markdown language, which gets converted by GitHub to html when viewed.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/index.md

cjmalchow, 2021-05-16:
Awesome, I'll see what I can put together! Looks like markdown formatting
is similar to HTML, so I shouldn't have an issue providing documentation in
that format.

I'll be in touch,

Cameron

jneilliii, 2021-05-16:
Awesome. Please base your changes on the rc branch, as that is currently being actively developed and may have updates not in the master branch.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/rc/wiki/index.md

cjmalchow, 2021-05-16:
While I'm at it, is there anything in particular you need tested on the rc
branch?

Cam

jneilliii, 2021-05-16:
Not really, I've been working through some bugs related to the new date locale stuff, but other than that I think it's solid.