jneilliii/OctoPrint-BedLevelVisualizer #430: [FR]: update mesh command

cmontgo, 2021-05-12:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

jneilliii, 2021-05-12:
Not quite sure I understand what you mean. If you mean to include in your start gcode? Need more details.

cmontgo, 2021-05-12:
So I've built a command in the Custom Commands that runs my bed levelling for me, and I was wondering if there is gcode (or other commands) I can add to that custom command to trigger a "Update Mesh" command (blue button in the main window)

jneilliii, 2021-05-12:
Oh yeah, All you have to do is include the `@BEDLEVELVISUALIZER` flag followed by the mesh report command (based on your firmware) in the commands being sent by a custom command button. If you share an example of your custom command I can possibly help more.

cmontgo, 2021-05-12:
So here is my script (Don't judge, I expect it has a glaring error somewhere)

> M104 S50;
> M140 S48;
> G28;
> M412 S0;
> M109 S60;
> M190 S48;
> M117 Leveling;
> G29;
> M851 Z-2.0;
> M420 L0 S1 Z5;
> M500;
> M73 P100;
> M412 S1;
> M117 Level Complete;

So I should just insert this in the second last step?
> @BEDLEVELVISUALIZER G29 Q1

jneilliii, 2021-05-12:
If M420 V reports your mesh data, then yes that would work, but needs on their own lines, like below.

```
M104 S50;
M140 S48;
G28;
M412 S0;
M109 S60;
M190 S48;
M117 Leveling;
G29;
M851 Z-2.0;
M420 L0 S1 Z5;
M500;
M73 P100;
M412 S1;
@BEDLEVELVISUALIZER
M420 V
M117 Level Complete;
```

cmontgo, 2021-05-12:
So the '@BEDLEVELVISUALIZER' tells it to update based on the following code?

jneilliii, 2021-05-12:
Yes, that flag is what tells the plugin to start reading in gcode commands coming back from your printer. It will continue collecting those gcode lines until an ok is received. That minimizes the constant watching of every command coming from your firmware to prevent possible performance issues. You could also use `M118 E1 BEDLEVELVISUALIZER` if your printer's firmware echoes the `BEDLEVELVISUALIZER` command back.

cmontgo, 2021-05-13:
I'll try that out, thanks!

closed 2021-05-23 by cmontgo