jneilliii/OctoPrint-BedLevelVisualizer #431: [BUG]: Y axis displays two values.

twild1990, 2021-05-15:
**Describe the bug**

Y axis values display two values at once.
**Expected behavior**

I would expect the visualizer to list 0 , 115, 230
**Debug Log**

**Screenshots**

[image]

**Firmware and Version**

Marlin 2.0.x-bugfix

jneilliii, 2021-05-15:
double check your printer profile inside octoprint. I think you have your rectangular dimensions wrong. notice the y max is wrong.

```
{'type': 'rectangular', 'x_min': 0, 'x_max': 220.0, 'y_min': 0, 'y_max': 200220.0, 'z_min': 0, 'z_max': 250.0}
```

twild1990, 2021-05-15:
Thank you. This corrected the issue. You've made my day that much better!

closed 2021-05-15 by jneilliii