jneilliii/OctoPrint-BedLevelVisualizer #432: [BUG]: 1.0.1-RC4 | Reported mesh is not translated to graph

hawkeyexp, 2021-05-15:
1.0.1 is working without any problems.

RC4:

Start data collect:
```
@BEDLEVELVISUALIZER
M420 V

```

Result:

```
[2021-05-15 17:43:06,543] DEBUG: mesh collection started
[2021-05-15 17:43:06,562] DEBUG: ['-0.190', '-0.071', '-0.012', '-0.019', '-0.063']
[2021-05-15 17:43:06,563] DEBUG: ['-0.176', '-0.048', '+0.007', '+0.002', '-0.021']
[2021-05-15 17:43:06,565] DEBUG: ['-0.173', '-0.041', '+0.006', '+0.008', '-0.011']
[2021-05-15 17:43:06,566] DEBUG: ['-0.171', '-0.036', '+0.010', '-0.001', '-0.025']
[2021-05-15 17:43:06,568] DEBUG: ['-0.142', '-0.014', '+0.036', '-0.001', '-0.031']
[2021-05-15 17:43:06,571] DEBUG: {'type': 'rectangular', 'x_min': 0, 'x_max': 220.0, 'y_min': 0, 'y_max': 220.0, 'z_min': 0, 'z_max': 210.0}
[2021-05-15 17:43:06,571] DEBUG: stopping mesh collection
[2021-05-15 17:43:06,572] DEBUG: Final mesh:
[2021-05-15 17:43:06,572] DEBUG: ['-0.142', '-0.014', '+0.036', '-0.001', '-0.031']
[2021-05-15 17:43:06,573] DEBUG: ['-0.171', '-0.036', '+0.010', '-0.001', '-0.025']
[2021-05-15 17:43:06,573] DEBUG: ['-0.173', '-0.041', '+0.006', '+0.008', '-0.011']
[2021-05-15 17:43:06,573] DEBUG: ['-0.176', '-0.048', '+0.007', '+0.002', '-0.021']
[2021-05-15 17:43:06,573] DEBUG: ['-0.190', '-0.071', '-0.012', '-0.019', '-0.063']
```

But graph won't update :-(

Not found any error in octoprint log or the debug log of the plugin.

jneilliii, 2021-05-15:
That's strange, as seems a duplicate of #424 which was resolved previously. Do you have ay other plugins that utilize plotly graphing? Just need to make sure there's not a conflict. There was another user that report a potential issue on the forum [here](https://community.octoprint.org/t/bed-level-visualizer/1383/182?u=jneilliii), which also appears to be using odd number of mesh points, which could be the underlying issue. 

Could you do me a favor and enable debug logging in the plugin, enable serial logging in OctoPrint's serial connection settings, restart OctoPrint and try to run the update mesh process again, and then supply both of those logs so I can reproduce with the virtual printer and your serial communication.

tablatronix, 2021-05-15:
I thought it was working, but it still seems to be intermittently failing still, I just tested it 3 times, it saved once

tablatronix, 2021-05-15:
```php
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.364 +0.274 +0.163 +0.079 +0.040
Recv:  1 +0.327 +0.232 +0.132 +0.050 +0.002
Recv:  2 +0.282 +0.159 +0.052 -0.040 -0.090
Recv:  3 +0.220 +0.094 -0.001 -0.097 -0.151
Recv:  4 +0.158 +0.050 -0.051 -0.135 -0.216
Recv: 
Recv: X:235.00 Y:211.00 Z:7.77 E:0.00 Count X:18800 Y:16880 Z:2985
Recv: ok
Send: M155 S3
Recv: ok
Send: M140 S0
Recv: ok
Send: M117 bed timed out
Recv: ok
Send: M140 S0
Recv: ok
Send: M117 bed timed out
Recv: ok
Send: M140 S0
Recv: ok
Send: M117 bed timed out
Recv: ok
Send: M140 S0
Recv: ok
Send: M117 bed timed out
Recv: ok
Send: M140 S0
Recv: ok
Send: M117 bed timed out
Recv: ok
Send: M140 S0
Recv: ok
Send: M117 bed timed out
Recv: ok
Send: M140 S0
Recv: ok
Send: M117 bed timed out
Recv: ok
Send: M140 S0
Recv: ok
Send: M117 bed timed out
Recv: ok
[...]
Send: G0 X118 Y118 F4000
Recv: ok
[...]
Send: M420 S1 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.364 +0.274 +0.163 +0.079 +0.040
Recv:  1 +0.327 +0.232 +0.132 +0.050 +0.002
Recv:  2 +0.282 +0.159 +0.052 -0.040 -0.090
Recv:  3 +0.220 +0.094 -0.001 -0.097 -0.151
Recv:  4 +0.158 +0.050 -0.051 -0.135 -0.216
Recv: 
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height OFF
Recv: ok
```

Exact same mesh second time and timestamp didnt update on screen

tablatronix, 2021-05-15:
```
Send: M420 S1 V
Recv: echo:Invalid mesh.
Recv: Error:Failed to enable Bed Leveling
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Failed to enable Bed Leveling
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height OFF
```
hmm

jneilliii, 2021-05-15:
Yeah, that looks like an issue on the printer side, not the plugin side.

jneilliii, 2021-05-15:
Oh I see. You reset EEPROM, which clears any stored meshes?

jneilliii, 2021-05-15:
Never mind, that's M502 not M503, my bad.

jneilliii, 2021-05-15:
@tablatronix I'm curious if your firmware is clearing mesh with G28? If `RESTORE_LEVELING_AFTER_G28` isn't enabled in firmware that can happen I think. If that is the case then maybe this plugin could help. https://plugins.octoprint.org/plugins/restorelevelingafterg28/

tablatronix, 2021-05-15:
Its enabled

tablatronix, 2021-05-15:
Mesh shows result, but timestamp does not, and if I refresh it goes back to the previous graph

[image]

[Archive.zip](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/6488188/Archive.zip)

tablatronix, 2021-05-15:
Hmm interesting, when I hover that new graph, its 3x3 not 4x4 ?

jneilliii, 2021-05-15:
Yeah, the data looks off that it's collected for some reason. Can you try with the relative z offset option disabled and z-limits set to -8,12?

hawkeyexp, 2021-05-15:
Sorry for the delay - had to wait for a print.  During prepare of loging i noticed an interesting thing:

i requested again the values from printer to check if the map is updated - no. To ensure it's a really visible difference i changed the z-offset and request a new run - no change.

But: i enabled all loggings and restart octoprint and hey the map was updated - ONLY the map not the timestring!
Seems one part is a refresh problem to load the updated map (brwoser refresh won't help).

Atached you will find the logs from a clean run of abl via your plugin.

https://cloud.hawknetwork.de:8443/s/GfmAJJPXJ3xnCHX

jneilliii, 2021-05-15:
Thanks guys, I'll look into this over the next week to see if I can figure out what's up. I appreciate all the feedback and logs. It will help to get to the bottom of it. I wonder if I messed something up with the internationalization of the date string that's displayed there. Do either of you get errors in your browser with the developer tools (F12) open on the console tab during the run of the update mesh process? 

Also, don't forget to disable serial logging as it can eat up your hard drive real quick with large prints.

hawkeyexp, 2021-05-15:
During a quick check there are no critical errors - only some warnings about stylesheet but this should be the problem.

Thx for your hard work - over all your plugins are great and essential for a nice octoprint experience :-)

tablatronix, 2021-05-15:
I think you are on to something!
[image]

jneilliii, 2021-05-16:
Could you guys do me a favor and see if setting the date locale option in settings `en-US` resolves the issue are not?

jneilliii, 2021-05-16:
Or your preferred date locale string of course, as described in the updated wiki description for the setting.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/rc/wiki/index.md#visualization

tablatronix, 2021-05-16:
hmm, why is my settings all wonky? I never even noticed locale, its like far far to the right I have to scroll..

[image]
[image]

jneilliii, 2021-05-16:
Something definitely wonky going on there with the way it's reading the data.

jneilliii, 2021-05-16:
try switching your limits and relative settings back to like before and see if that gets it back to normal.

tablatronix, 2021-05-16:
I could not get it back to anything, I erased data and am trying fresh

tablatronix, 2021-05-16:
Side note, when not connected to printer this shows up, seems wrong, should say not available or something ?

[image]

tablatronix, 2021-05-16:
ok back in business, strange..
I will keep testing and see if it works a few times, not sure why I am getting timeouts all of a sudden, I lost my gcode so maybe I had a fix in there...

[image]

tablatronix, 2021-05-16:
So far so good, It is actually useful again, not sure what happened, maybe settings got corrupted?

[image]

Not sure what the com errors are from, seems octoprint has a timeout while waiting for the mesh to be returned, marlin must be taking longer unrelated to plugin probably.

jneilliii, 2021-05-16:
well I can say that if the js side fails like it was because of the locale it might have blocked the save of the mesh data to storage, which is why it wouldn't graph as well. So the issue is a blank date locale causing that. I will have to compare that value against a blank string and not send the locale if it is blank, that should fix the auto-detect stuff. In the interim I have defaulted the rc branch to `en-US` so it doesn't cause errors.

hawkeyexp, 2021-05-16:
great that you could identify the reason

jneilliii, 2021-05-16:
I've just pushed 1.1.0rc5 that should resolve a blank date string for auto-detection from browser, and hopefully finally resolve this issue, if you two can verify that would be great. Interested in knowing where @tablatronix is with his weird firmware responses, I think I noticed you in the Discord server asking about it recently.

hawkeyexp, 2021-05-16:
i pulled it today directly from git and it is working on  my side :-)

tablatronix, 2021-05-16:
it was odd that it just manifested when I was debugging this. I did not do any updates recently that I know of, but I will double check, maybe this is a new octoprint thing..

But marlin is giving me a delay with no busy after G29 and octoprint is giving me idle timeouts, I had to increase the comm and idle timeouts ,not sure maybe enabling serial log changed something in octoprint settings, or some of these timeouts got initialized when I saved something in settings.

closed 2022-01-01 by jneilliii