jneilliii/OctoPrint-BedLevelVisualizer #433: [BUG]: Mesh is invalid

kfrancis, 2021-05-17:
**Describe the bug**
I can't complete a UBL level process with the plugin as once it finishes leveling and reports, the plugin/marlin bugs out. I can use UBL just fine through the printer and view/use the stored mesh but I can't do this through the plugin without a halt.

**Expected behavior**
The plugin should finish the leveling process, report the mesh and complete.

**Debug Log**
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/6496537/plugin_bedlevelvisualizer_debug.log)

**Screenshots**
[image]
[image]
[image]

**Firmware and Version**
- browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/90.0.4430.212 Safari/537.36 Edg/90.0.818.62
- connectivity.connection_check: 1.1.1.1:53
- connectivity.connection_ok: true
- connectivity.enabled: true
- connectivity.online: true
- connectivity.resolution_check: octoprint.org
- connectivity.resolution_ok: true
- env.hardware.cores: 4
- env.hardware.freq: 1500
- env.hardware.ram: 4041732096
- env.os.bits: 32
- env.os.id: linux
- env.os.platform: linux
- env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.1
- env.plugins.pi_support.octopi_version: 0.17.0
- env.plugins.pi_support.throttle_state: 0x0
- env.python.pip: 20.2.3
- env.python.version: 3.7.3
- env.python.virtualenv: true
- octoprint.safe_mode: false
- octoprint.version: 1.6.1
- printer.firmware: Marlin 2.0.8 (May 8 2021 18:47:05)
- systeminfo.generator: systemapi

jneilliii, 2021-05-17:
My plugin doesn't have M112 embedded in it, that could be causing part of the issue here. Do you have the Thermal Runaway plugin installed? Apparently the last update is bugged and causing a few issues for people.

kfrancis, 2021-05-21:
Hmm, plugin? Like octoprint plugin? Or do you mean something in Marlin?

jneilliii, 2021-05-21:
Yeah, the OctoPrint plugin, Thermal Runaway.

kfrancis, 2021-05-23:
Hmm, don't have that installed. Question, could it be the number of significant digits? I was using `SERIAL_FLOAT_PRECISION 4`

jneilliii, 2021-05-23:
Not sure what that does, but the plugin doesn't care about the precision of the data returned. It uses the regular expression 
```
(\+?-?\d*\.\d*)
``` 
to match pretty much any number.

jneilliii, 2021-05-23:
have you tried running `G29 P3` multiple times?

plucker, 2021-06-08:
I have this issue as well, exactly the same with a 15x15 mesh, bltouch on an ender 3 v2 with 4.2.7 motherboard and Marlin release v1.3.4 from here https://github.com/Jyers/Marlin/releases.  It goes through the 15x15 probes then fails with the same error in Bed Visualizer

plucker, 2021-06-08:
I've just re-installed Bed Visualizer and this time it has worked with a 15x15 mesh

kfrancis, 2021-06-16:
I've tried re-installing, nothing. I've tried running G29 P3 a number of times, nothing.

jneilliii, 2021-06-16:
If you want help, enable bedbug logging, restart octoprint ans try again. If not good, provide logs.

kfrancis, 2021-06-17:
I did, on the initial report .. do you want new ones?

jneilliii, 2021-06-17:
@kfrancis your specific issue was the firmware error causing M112 to be sent to the printer, which my plugin doesn't control. By chance are you using jyers firmware? I've learned recently that OctoPrint needs higher timeout values with that firmware in the core settings. 

https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings#error-handling

closed 2021-07-02 by kfrancis

mlemanczyk, 2024-11-22:
I'm facing the same issue with my CR-10 v2 on Unified Firmware from TH3D. I've ran it several times and it always finished with the same error. Interestingly, on the last attempt I've commented out the line "@BEDLEVELVISUALIZER	; tell the plugin to watch for reported mesh" and now it finished and I can see the mesh.

G28       ; home all axes
M420 S0   ; Turning off bed leveling while probing, if firmware is set
          ; to restore after G28
M155 S30  ; reduce temacperature reporting rate to reduce output pollution
M190 S60  ; (optional) wait for the bed to get up to temperature
G29 P1    ; automatically populate mesh with all reachable points
G29 P3    ; infer the rest of the mesh values
G29 P3    ; infer the rest of the mesh values again
;@BEDLEVELVISUALIZER	; tell the plugin to watch for reported mesh
M420 S1 V ; enabled leveling and report the new mesh
G29 S0    ; Save UBL mesh points to slot 0 (EEPROM).
G29 F 10.0 ; Set Fade Height for correction at 10.0 mm.
G29 A     ; Activate the UBL System.
M500      ; save the current setup to EEPROM
M155 S3   ; reset temperature reporting
M140 S0   ; cooling down the bed