jneilliii/OctoPrint-BedLevelVisualizer #435: [BUG]: plugin missing after install

whitewhidow, 2021-05-19:
**Describe the bug**
After installing (reports success) and rebooting octoprint, the plugin is not installed (also listed as not installed in plugin list)

**Expected behavior**
the plugin to be installed

**Debug Log**
missing

**Firmware and Version**
octoprint Version 1.6.1

jneilliii, 2021-05-19:
Did you migrate your Python 2 OctoPrint to Python 3 using the upgrade script?

jneilliii, 2021-05-19:
If that is the case, check out this.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer#known-issues

whitewhidow, 2021-05-19:
regular octoprint reported
`ImportError: libf77blas.so.3: cannot open shared object file: No such file or directory`
when trying to launch the plugin

fixed with `sudo apt-get install libatlas-base-dev`

closed 2021-05-19 by whitewhidow