jneilliii/OctoPrint-BedLevelVisualizer #441: You must be logged in

TimmyOl, 2021-05-28:
I've never succeded in getting this to work but i really want to, i get the "You must be logged in" with a warning triangle, logged in to what? and how do i log in?

It doesn't get any mesh and my guess is that this is the reason since i have the gcode set up? as soon as i click the bed visualizer button it just tries to get a mesh but doesn't run the gcode and below it says: " You must be logged in.
In order for this plugin to work properly, you must configure the GCODE commands necessary to send back a Bed Topography Report in Settings."

below that there is 2 buttons that doesn't do anything, one with an "i" and one with a cog i guess should take me to the settings page?

jneilliii, 2021-05-28:
Enable debug logging, restart OctoPrint, try again. Once it fails (assuming it does) then grab the files octoprint.log and plugin_bedlevelvisualizer_debug.log from the logging section of OctoPrint and drag them into a comment here to upload. Also please share the gcode settings you are using, the firmware variant and version, and Auto Bed Leveling type configured in your firmware.

TimmyOl, 2021-05-28:
> Enable debug logging, restart OctoPrint, try again. Once it fails (assuming it does) then grab the files octoprint.log and plugin_bedlevelvisualizer_debug.log from the logging section of OctoPrint [...]

Okej, but No it doesnt fail, just spinning the wheel and "Please wait, retrieving mest"
[image]

Sorry for the bad image, ill check for the logs tomorrow

jneilliii, 2021-05-28:
that looks strange to me, the logs, as well as check the developer tools (F12) in your browser and look for any red errors on the console tab on initial page load and while it's doing that process. The above looks like a broken UI, and there could be a conflict with another plugin as well. also share a screenshot of your gcode commands you're using for reference as well.

closed 2021-06-19 by github-actions[bot]

felixlandmeyer, 2026-10-11:
This is probably not solving the described issue, but I got a similar issue because my octoprint account was set to 'admin' only and not 'operator'. Setting it to 'operator' too solved the problem for me.