jneilliii/OctoPrint-BedLevelVisualizer #443: Wont create mesh...

Jhanson641501, 2021-06-04:
Hello, I apologize ahead of time if this has been solved before... I understand the printers and the mechanical side of things but when it comes to coding... I dont stand a chance. 

I am using version 1.0.1. Just installed a couple days ago. Should be the newest version.
This is on a Ender 3 Pro, running a Skr E3 turbo with Teaching Tech example firmware for it. 

Ive ran the debug and got the following....

[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/6595061/plugin_bedlevelvisualizer_debug.log)

I hope that uploaded correctly. 

This is the gcode im running, its from the example gcode thats provided...

M140 S60 ; starting by heating the bed for nominal mesh accuracy
M117 Homing all axes ; send message to printer display
G28      ; home all axes
M117 Heating the bed ; send message to printer display
M190 S60 ; waiting until the bed is fully warmed up
M300 S1000 P500 ; chirp to indicate bed mesh levels is initializing
M117 Creating the bed mesh levels ; send message to printer display
M155 S30 ; reduce temperature reporting rate to reduce output pollution
@BEDLEVELVISUALIZER	; tell the plugin to watch for reported mesh
G29 T	   ; run bilinear probing
M155 S3  ; reset temperature reporting
M140 S0 ; cooling down the bed
M300 S440 P200 ; make calibration completed tones
M300 S660 P250
M300 S880 P300
M117 Bed mesh levels completed ; send message to printer display

Current mesh data is;

Mesh z Height in:
--
0 | 0.122 | 0.053 | 0.070
118 | 0.190 | 0.050 | 0.015
235 | 0.285 | 0.050 | -0.047

Timeout is at 300secs.
Save mesh is on.

I've tried to work through a couple other peoples issues to see if they would fix mine, but it still wont create the graph.

I hope I added everything that is needed. 
Any and all help will be appreciated, I'm sure its something simple I'm missing. lol

jneilliii, 2021-06-04:
use the option to remove row labels and try again.

jneilliii, 2021-06-04:
scratch that, it looks like the data is collected properly actually, I assume that mesh z height you listed there is purely a copy/paste from the mesh data?

some browser extensions actually block the rendering of the mesh. I've heard of this from people using brave browser IIRC. try doing the update mesh process while running your browser in incognito mode to prevent any potential extensions from interfering as well.

Jhanson641501, 2021-06-04:
You got it! I was using chrome and as soon as I entered incognito mode, it worked right away. I knew it had to be something simple. You are the best! Thank you so much!!

jneilliii, 2021-06-04:
So the trick is to figure out which extension is blocking webgl and enable it for your octoprint instance and it should in theory work in regular mode.

closed 2021-06-04 by jneilliii