jneilliii/OctoPrint-BedLevelVisualizer #445: [BUG]: Restricted user as access to all seetings

Danafra, 2021-06-08:
Normally, a user added in Octoprint as in "Operator" does not have access to Octoprint's settings. However, if the user access's the settings for Bed Visualizer, they have access to all settings.

This is using Bed Visualizer 1.0.1 and Octoprint 1.6.1.

[image]
[image]
[image]

jneilliii, 2021-06-08:
Quick question, that allows them to open the settings, but are they allowed to save changes? If so that almost seems like a higher end bug that should be reported to core OctoPrint as well. I will definitely adjust this plugin to not show the button unless the user has the proper permissions.

jneilliii, 2021-06-20:
fixed by above commit and will be included in the next release.

closed 2022-01-01 by jneilliii