jneilliii/OctoPrint-BedLevelVisualizer #446: [FR]: Optional allow mesh reporting during print job

hawkeyexp, 2021-06-13:
Would it be possible to allow to catch the mesh points during an print job?

Background:
a while ago i added the leveling by a bltouch to my startcode - so the mesh is updated on every run (i added a magnetic print bed so it's not sure that leveling will fit 100% after remove print plate and add back).
It would be nice if i can check the leveling during the current printjob.

Currently the plugin is blocked during run (which is ok for normal usecases :-) and adding the capture command to startcode is useless :-)

jneilliii, 2021-06-13:
You can do this now you just need to add the command `@BEDLEVELVISUALIZER` just prior to the G29 command you use in your start gcode. You can also use `M118 BEDLEVELVISUALIZER` to get more "realtime" collection from firmware and the plugin doesn't have to work as hard. Then when you go to the tab it will render the latest collected mesh.

jneilliii, 2021-06-13:
That's assuming you're using the option to save mesh.

hawkeyexp, 2021-06-13:
i will check it again - i was sure i was not working but let me try...

hawkeyexp, 2021-06-13:
it works - seems on last try something wen't wrong in my side - sorry

closed 2021-06-13 by hawkeyexp