jneilliii/OctoPrint-BedLevelVisualizer #448: [BUG]: Empty image exported upon completed probing.

mano1979, 2021-06-20:
**Describe the bug**
After doing a bed levelling, an image is exported which only has the legenda in it but not the actual mesh map.

**Expected behavior**
Image WITH the mesh map...

**Debug Log**
NaN

**Screenshots**
[image]

**Firmware and Version**
Marlin 1.1.8

jneilliii, 2021-06-20:
enable the option for using relative z offset, I think your mesh is outside the limits of -2,2 which is what's rendered in view.

mano1979, 2021-06-20:
I don't think I'm outside the limits. it is rendered fine in the web interface. just not in the image that is being exported.
[image]

mano1979, 2021-06-20:
I tried to set the z offset to relative but no difference at all.

jneilliii, 2021-06-20:
are you using the camera button to save the png file or the option to automatically save the snapshot when the mesh is rendered?

mano1979, 2021-06-20:
The last. Automatically save a snapshot

jneilliii, 2021-06-20:
Interesting, just tested this and it's working for me. Does the camera button do the same thing for you? Also, what browser are you using?

mano1979, 2021-06-20:
The camera button does work. I get a complete image with that.
I use Safari browser on Mac

jneilliii, 2021-06-20:
This could be a bug with plotly directly. I have just pushed version 1.1.0rc8 with the latest plotly release to see if it might fix this. If you change the plugin's release channel to `Release Candidate` in OctoPrint's Software Update settings and upgrade when prompted you can see if that's make any difference for you.

mano1979, 2021-06-20:
I updated the plugin but nothing changed. Let me know if you need me to test anything else or if you need som kind of log.

jneilliii, 2021-06-20:
It's strange that it's happening actually. I don't have access to a mac, so tricky to debug. Can you check safari's developer tools to see if there are any errors during the process. I'm just using the built in functions of plotly to download the file so a little worried I won't be able to fix directly.

mano1979, 2021-06-20:
Yes sure I can do that later. I have a print running at the moment. And it's quite late here, so I have to sleep soon. if I can't make it tonight, I will do so tomorrow.

mano1979, 2021-06-21:
Ok I copied everything, I didn't exactly know what you needed. But have look at the last line. There is an error there.

`< <div class="page-container" id="page-container-main" style data-bind="visible: visible">…</div>
[Log] ["@BEDLEVELVISUALIZER", "M140 S60 ; starting by heating the bed for nominal mesh accuracy", "M117 Homing all axes ; send message to printer display", "G28      ; home all axes", "M117 Heating the bed ; send message to printer display", "M190 S60 ; waiting until the bed is fully warmed up", "M300 S1000 P500 ; chirp to indicate bed mesh levels is initializing", "M117 Creating the bed mesh levels ; send message to printer display", "M155 S30 ; reduce temperature reporting rate to reduce output pollution", "@BEDLEVELVISUALIZER\t; tell the plugin to watch for reported mesh", …] (17) (packed_plugins.js, line 7458)
[Log] Array (7) (packed_plugins.js, line 7271)
0 [0, 0.005, -0.015, -0.015, 0.032, 0.045, -0.02] (7)
1 [-0.038, -0.073, -0.045, -0.01, 0, 0.05, 0.027] (7)
2 [0.005, 0.002, -0.04, -0.073, -0.025, -0.058, -0.088] (7)
3 [0.03, -0.033, -0.033, -0.078, -0.025, -0.018, -0.015] (7)
4 [-0.045, -0.015, -0.05, -0.075, -0.033, -0.065, -0.06] (7)
5 [-0.068, -0.07, -0.043, -0.065, -0.095, -0.103, -0.14] (7)
6 [-0.145, -0.04, -0.038, -0.075, -0.073, -0.15, -0.278] (7)

Array Prototype

concat(value, ...)

constructor: function()

arguments: TypeError: 'arguments', 'callee', and 'caller' cannot be accessed in this context.

caller: TypeError: 'arguments', 'callee', and 'caller' cannot be accessed in this context.

from(arrayLike, [mapFunction], [thisArg])

isArray(object)

length: 1

name: "Array"

of([...values])

prototype: Array (0)

concat()

constructor: function()

arguments: TypeError: 'arguments', 'callee', and 'caller' cannot be accessed in this context.

caller: TypeError: 'arguments', 'callee', and 'caller' cannot be accessed in this context.

from(arrayLike, [mapFunction], [thisArg])

isArray(object)

length: 1

name: "Array"

of([...values])

prototype: Array (0)

concat()

constructor: function()

arguments: TypeError: 'arguments', 'callee', and 'caller' cannot be accessed in this context.

caller: TypeError: 'arguments', 'callee', and 'caller' cannot be accessed in this context.

from(arrayLike, [mapFunction], [thisArg])

isArray(object)

length: 1

name: "Array"
[... 971 lines cut ...]
some(callback, [thisArg])

sort([compareFunction])

splice(startIndex, [deleteCount=0], ...itemsToAdd)

toLocaleString()

toString()

unshift()

values()

Symbol(Symbol.iterator)()

Symbol(Symbol.unscopables): {copyWithin: true, entries: true, fill: true, find: true, findIndex: true, …}

Object Prototype

[Log] Back [um]:-16 (packed_plugins.js, line 7411)
[Log] Front [um]:-68 (packed_plugins.js, line 7412)
[Log] Left [um]:-29 (packed_plugins.js, line 7413)
[Log] Right [um]:-64 (packed_plugins.js, line 7414)
[Error] Failed to load resource: Frame load interrupted (data:application/octet-stream;base64,iVB…mCC, line 0)`

jneilliii, 2021-06-21:
I've commented in the above issue looking for help. The same error there at the end is mentioned in that issue and is related to Safari.

mano1979, 2021-06-21:
Ah ok. let's hope someone knows a fix. I don't like other browsers. thanks for looking into it.

jneilliii, 2021-07-24:
There were some safari fixes release in the latest plotly version and I have updated in version 1.1.0rc9. Please report back if that fixed your issue or not 🤞.

mano1979, 2021-07-24:
I updated and tested it. Unfortunately no difference. Still no graph, only the legenda.

jneilliii, 2021-07-24:
I was hopeful, sorry it's still broken.

mano1979, 2021-07-24:
That's ok. Thank you for looking into it. Let me know if i need to test something.

rlreamy, 2022-12-05:
I am seeing a very similar issue with Safari and the latest version of plotly.js. You can recreate the bug by going to https://atlas.kpmp.org/explorer, then click on the first link in the table at the right with a red asterisk, lastly search for a gene (APOL1 will work). You will see the graph on the right spin for a few moments while it retrieves the data. Once that graph is generated, when you export the plot with the camera, the first export gives you only the x/y axes and the legend, the second time you export, you will get the graph as well.