jneilliii/OctoPrint-BedLevelVisualizer #449: [BUG]: Octoprint startup stops parsing bedlevelvisualizer

6ffm70, 2021-06-28:
**Describe the bug**
Starting Octoprint stops at `Parsing plugin metadata for bedlevelvisualizer from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer/__init__.py`
Octoprint will not be executed. This is reproduceable.
Starting Octoprint in safemode works.
After uninstalling only the BedLevelVisualizer plugin (including settings), Octoprint will start without issues.
When reinstalling BedLevelVisualizer, Octoprint will freeze, just after the installation dialogue confirms the install to be finished.

**Expected behavior**
The presence of BedLevelVisualizer should not stop Octoprint, or interrupt its start-up.

**Context**
I used BedLevelVisualizer in this instance w/o problems before. The only thing I changed prior to the unexpected behaviour was to `$ sudo apt update
$ sudo apt upgrade -y` and restart the Raspi. Note: the first reproduction log below ends in a _Warning_ about a version mismatch.

**Debug Log**
Not from the plugin, but from Octoprint startup:
1. reproduction

```
pi@60594:~/oprint/bin $ ./octoprint 
Starting the server via "octoprint" is deprecated, please use "octoprint serve" from now on.
2021-06-28 15:38:34,947 - octoprint.startup - INFO - ******************************************************************************
2021-06-28 15:38:34,949 - octoprint.startup - INFO - Starting OctoPrint 1.5.3
2021-06-28 15:38:34,950 - octoprint.startup - INFO - ******************************************************************************
2021-06-28 15:38:35,293 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2021-06-28 15:38:35,351 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is working
2021-06-28 15:38:35,351 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2021-06-28 15:38:35,483 - octoprint.startup - DEBUG - Blacklisted plugin: roomtemp
2021-06-28 15:38:35,485 - octoprint.startup - DEBUG - Blacklisted plugin: GcodeEditor, versions: 0.1.1, 0.2.0, 0.2.1, 0.2.2, 0.2.3, 0.2.4, 0.2.5, 0.2.6, 0.2.8
2021-06-28 15:38:35,488 - octoprint.startup - DEBUG - Blacklisted plugin: gcodeleveling, versions: 0.1.0, 0.1.1
2021-06-28 15:38:35,494 - octoprint.startup - INFO - Blacklist processing done, adding 12 blacklisted plugin versions: roomtemp (any), GcodeEditor (0.1.1), GcodeEditor (0.2.0), GcodeEditor (0.2.1), GcodeEditor (0.2.2), GcodeEditor (0.2.3), GcodeEditor (0.2.4), GcodeEditor (0.2.5), GcodeEditor (0.2.6), GcodeEditor (0.2.8), gcodeleveling (0.1.0), gcodeleveling (0.1.1)
2021-06-28 15:38:35,515 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2021-06-28 15:38:35,517 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for pi_support from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support/__init__.py
2021-06-28 15:38:35,527 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for discovery from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery/__init__.py
2021-06-28 15:38:35,563 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for announcements from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements/__init__.py
2021-06-28 15:38:35,871 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for virtual_printer from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer/__init__.py
2021-06-28 15:38:35,877 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for appkeys from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys/__init__.py
2021-06-28 15:38:35,889 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for tracking from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking/__init__.py
2021-06-28 15:38:35,902 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for gcodeviewer from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/gcodeviewer/__init__.py
2021-06-28 15:38:35,906 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for pluginmanager from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py
2021-06-28 15:38:35,959 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for backup from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup/__init__.py
2021-06-28 15:38:35,985 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for action_command_notification from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_notification/__init__.py
2021-06-28 15:38:35,990 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for errortracking from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/errortracking/__init__.py
2021-06-28 15:38:35,996 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for softwareupdate from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py
2021-06-28 15:38:36,090 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for action_command_prompt from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt/__init__.py
2021-06-28 15:38:36,097 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for corewizard from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard/__init__.py
2021-06-28 15:38:36,103 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for logging from AST of /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging/__init__.py
2021-06-28 15:38:36,194 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for touchui from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui/__init__.py
2021-06-28 15:38:36,204 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for prusaslicerthumbnails from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_prusaslicerthumbnails/__init__.py
2021-06-28 15:38:36,226 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for signalclirestapi from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_signalclirestapi/__init__.py
2021-06-28 15:38:36,243 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for PrintJobHistory from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py
2021-06-28 15:38:36,323 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for preheat from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_preheat/__init__.py
2021-06-28 15:38:36,339 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for widescreen from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_widescreen/__init__.py
2021-06-28 15:38:36,347 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for tempsgraph from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tempsgraph/__init__.py
2021-06-28 15:38:36,354 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for taborder from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_taborder/__init__.py
2021-06-28 15:38:36,365 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for SpoolManager from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_SpoolManager/__init__.py
2021-06-28 15:38:36,404 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for smartpreheat from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_smartpreheat/__init__.py
2021-06-28 15:38:36,413 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for signalnotifier from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_signalnotifier/__init__.py
2021-06-28 15:38:36,425 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for resource_monitor from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_resource_monitor/__init__.py
2021-06-28 15:38:36,483 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for PrusaLevelingGuide from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrusaLevelingGuide/__init__.py
[... cut ...]
2021-06-28 15:39:52,784 - octoprint.plugin.core - INFO - Plugin Continuous Print (1.0.3) is disabled.
2021-06-28 15:39:52,790 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for bedlevelvisualizer from AST of /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer/__init__.py
Segmentation fault
```

**Screenshots**

**Firmware and Version**
|          model: Raspberry Pi 4 Model B Rev 1.1
OctoPrint version : 1.5.3
OctoPi version    : 0.17.0
|  python:
|      pip: 19.3.1
|      version: 2.7.16
Bed_Visualizer-1.0.1.

jneilliii, 2021-06-28:
This is more than likely an upstream issue with numpy compiling on python 2. I would recommend you to upgrade your python environment used by OctoPrint to version 3 using the upgrade script linked below.

https://github.com/cp2004/Octoprint-Upgrade-To-Py3

jneilliii, 2021-06-28:
or alternatively, use the current release candidate branch for the plugin in OctoPrint's software update settings, switch the release channel to that and update when prompted. That version removed the numpy requirement.

6ffm70, 2021-07-13:
> or alternatively, use the current release candidate branch for the plugin in OctoPrint's software update settings, switch the release channel to that and update when prompted. That version removed the numpy requirement.

Worked for me, thank you!