jneilliii/OctoPrint-BedLevelVisualizer #450: [FR]: front,left,right,back labels

legend069, 2021-07-03:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

hello, i think it would be really neat to add custom labels to the UI
for example where it has the X axis being able to add the "front,Rear,Left,Right" would be nice

see image for example
https://imgur.com/O2qfCBM

**Describe alternatives you've considered**

modifying  your code :)

**Additional context**

see image for example
https://imgur.com/O2qfCBM

jneilliii, 2021-10-08:
The camera position option allows you to change the initial viewpoint.

aalku, 2021-10-11:
True but I am not able to understand how to place the camera where I want. It's totally confusing for me.

bratwiz, 2021-11-12:
Hello,

I would very much like to see "Front", "Rear", "Left", "Right" labels also. It can be confusing looking at the screen and wondering whether the top of the chart indicates I should be using the front or the back adjustment wheels.

I just discovered (from reading in this thread) the feature / ability to move the nozzle directly using the tools in the graph. But I don't see how to take a reading at that location..??

Thank you for your hard work on this plugin !!!

julesgilson, 2023-07-19:
> I think I figured out a way that you can determine this camera setting until I can get around to adding the feature to save the camera position in a future update. If you position the camera the way you want it, then open your browser's developer tools and run the command below in the console tab it should return the 3 values to enter.
> 
> ```
> $('#bedlevelvisualizergraph')[0].layout.scene.camera.eye;
> ```

alternatively check this object:

`$('#bedlevelvisualizergraph')[0]._fullLayout.scene.camera.eye`

Jules.