jneilliii/OctoPrint-BedLevelVisualizer #451: Won't install

Spydyr, 2021-07-04:
7 hours sitting installing and nothing. It will not install for any reason.

jneilliii, 2021-07-04:
Well, no logs, errors or any additional information other than "it doesn't install" is not helpful in debugging your problem. My best guess is it's locking up on compiling numpy. You could try my release candidate version that doesn't have numpy requirement by copying/ pasting the url below into plugin manager > get more > ...from URL.

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/refs/tags/1.1.0rc8.zip
```

Spydyr, 2021-07-04:
Could not get any logs as it was in the installing from plugin manager for 7 hours.

Spydyr, 2021-07-04:
Also numpy was installed before I tried installing visualizer because I thought that was a problem.

Spydyr, 2021-07-04:
Started install now from your new link which is different from plugin manager. Time is 9:01 AM CDT.

closed 2021-07-04 by Spydyr

jneilliii, 2021-07-04:
And I assume that one worked for you?

Spydyr, 2021-07-04:
Worked on one that has 2 RPi brand power supplies attached to it but will not work on my other that is powered by a 5v 70A supply. Suspecting I have 2 bad RPi as they show low voltage at 5.15v.