jneilliii/OctoPrint-BedLevelVisualizer #455: with latest bed visual plugin, I lost the graph. Is there a way to revert to the previous rev of the plugin?

steveiebbb, 2021-08-05:
I checked the known issues, but it seems like this was an issue corrected in the last release? Maybe I am not seeing the release notes correctly. I apologize if this is a duplicate

steveiebbb, 2021-08-05:
update. I had been running 2.0.9.0. Visualization works. Recompiled for 2.0.9.1 nightly bugfix, Visualization works. Went to latest Visualization plugin, Visualization fails, Revert to previous 2.0.9.0. Visualization works again. Checked and I have this, which SHOULD be correct?

//#define RESTORE_LEVELING_AFTER_G28
#define ENABLE_LEVELING_AFTER_G28

jneilliii, 2021-08-05:
You probably also want restore as g28 will null out mesh data.

jneilliii, 2021-08-05:
But it's hard to say without seeing the commands you are using or having any logs to look at...

Enable debug logging, restart octoprint, try again. If fails share your logs, both octoprint.log and plugin_bedlevelvisualizer_debug.log.

steveiebbb, 2021-08-05:
I saw where to enable serial and bed logging, but where does it park the log files?
Thanks for the quick reply!

jneilliii, 2021-08-05:
In OctoPrint's settings under the logging section.

steveiebbb, 2021-08-06:
ok, I will log both the working 2.0.9.0 build results and the nonworking 2.0.9.1 nightly build results, both using the new plugin.it may not be until tomorrow. again, thanks for the help. Im just glad I have a working combo but would love to get the new build working

steveiebbb, 2021-08-06:
It looks like it attempts another mesh collection and then just sits there...

jneilliii, 2021-08-06:
if you attempted to share a file via email, that won't work. You have to open github and drag the files into the comment to attach.

steveiebbb, 2021-08-06:
[fails_octoprint.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/6947106/fails_octoprint.log)
[fails_plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/6947108/fails_plugin_bedlevelvisualizer_debug.log)
[serial.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/6947111/serial.log)

hope this worked

jneilliii, 2021-08-06:
perfect, I'll try to take a look this weekend.

steveiebbb, 2021-08-07:
whenever you can! i am not dead in the water by any means. IS there a way to roll back an update to the previous state, btw?

jneilliii, 2021-08-07:
You can copy/paste this URL in the plugin manager > get more > ...from URL. 

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/refs/tags/1.0.1.zip
```

jneilliii, 2021-08-07:
Thanks, I think I see the error...stand by.

jneilliii, 2021-08-07:
I take that back. The error I was seeing was in the previous version in your octoprint.log about a missing file.

jneilliii, 2021-08-07:
ok, I can see from the plugin's log that it is collecting the data, so the issue must be on the browser side. Could you check the console tab in your browser's developer tools (F12 on your keyboard) and let me know if you see any errors that come up during initial page load and during the mesh collection process please?

steveiebbb, 2021-08-07:
IS there a way to start fresh logs, or do you not need me to?

steveiebbb, 2021-08-07:
These show upon page load...

:5000/#term:1 The resource http://localhost:5000/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.:5000/#term:1 The resource http://localhost:5000/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.:5000/#term:1 The resource http://localhost:5000/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.:5000/#term:1 The resource http://localhost:5000/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.:5000/#term:1 The resource http://localhost:5000/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.:5000/#term:1 The resource http://localhost:5000/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.

This when I retrieve mesh...
   
   
   - ***@***.***", "G28", "G29"]
   
   - ***@***.***"
   - 1: "G28"
   - 2: "G29"
   - length: 3

steveiebbb, 2021-08-07:
I put the old 2.0.9.0 compile back and everything looked same except upon succesful mesh, this also appeared in the console...

5) [Array(5), Array(5), Array(5), Array(5), Array(5)]packed_core.js?285422d8:837 ConfiguredLoggers has not changed. Not saving.packed_plugins.js?550e3511:6383 Back [um]:3packed_plugins.js?550e3511:6384 Front [um]:30packed_plugins.js?550e3511:6385 Left [um]:17packed_plugins.js?550e3511:6386 Right [um]:19

jneilliii, 2021-08-07:
I'm stumped at this point. The only thing I see that might be related is the error in octoprint.log

```
2021-08-05 09:39:31,987 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Traceback (most recent call last):
  File "c:\octoprint\venv\lib\site-packages\octoprint\util\comm.py", line 3860, in _readline
    ret = self._serial.readline()
  File "c:\octoprint\venv\lib\site-packages\octoprint\util\comm.py", line 6488, in readline
    c = self.read(1)
  File "c:\octoprint\venv\lib\site-packages\serial\serialwin32.py", line 295, in read
    raise SerialException("GetOverlappedResult failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: GetOverlappedResult failed (PermissionError(13, 'Access is denied.', None, 5))
```

None of the console errors that you shared indicate a plugin error on the js side, or  missing dependency. 

If you stop OctoPrint and put this in the address bar of file explorer `%appdata%\OctoPrint\logs\` and delete the 3 log files, Then when you start OctoPrint back up they will be recreated. 

At this point, the only other thing I can think to do is uninstall the plugin with the option to clear the data and start over from scratch.

steveiebbb, 2021-08-07:
you shoed me this URL...https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/refs/tags/1.0.1.zip

will it revert me back to the previous plugin?

jneilliii, 2021-08-08:
Yes, that will downgrade the version to the last released one. You will get prompted to upgrade again, just click the ignore button.

steveiebbb, 2021-08-08:
got this...

ERROR: Command errored out with exit status 1:command: 'c:\octoprint\venv\scripts\python.exe' 'c:\octoprint\venv\lib\site-packages\pip\_vendor\pep517\_in_process.py' prepare_metadata_for_build_wheel 'C:\Users\owner\AppData\Local\Temp\tmpejlhstnm'cwd: C:\Users\owner\AppData\Local\Temp\pip-install-7y3db7s3\numpyComplete output (192 lines):Running from numpy source directory.setup.py:470: UserWarning: Unrecognized setuptools command, proceeding with generating Cython sources and expanding templatesrun_build = parse_setuppy_commands()Processing numpy/random\_bounded_integers.pxd.inProcessing numpy/random\bit_generator.pyxProcessing numpy/random\mtrand.pyxProcessing numpy/random\_bounded_integers.pyx.inProcessing numpy/random\_common.pyxProcessing numpy/random\_generator.pyxProcessing numpy/random\_mt19937.pyxProcessing numpy/random\_pcg64.pyxProcessing numpy/random\_philox.pyxProcessing numpy/random\_sfc64.pyxCythonizing sourcesblas_opt_info:blas_mkl_info:No module named 'numpy.distutils._msvccompiler' in numpy.distutils; trying from distutilscustomize MSVCCompilerlibraries mkl_rt not found in ['c:\\octoprint\\venv\\lib', 'C:\\']NOT AVAILABLEblis_info:libraries blis not found in ['c:\\octoprint\\venv\\lib', 'C:\\']NOT AVAILABLEopenblas_info:libraries openblas not found in ['c:\\octoprint\\venv\\lib', 'C:\\']get_default_fcompiler: matching types: '['gnu', 'intelv', 'absoft', 'compaqv', 'intelev', 'gnu95', 'g95', 'intelvem', 'intelem', 'flang']'customize GnuFCompilerCould not locate executable g77Could not locate executable f77customize IntelVisualFCompilerCould not locate executable ifortCould not locate executable iflcustomize AbsoftFCompilerCould not locate executable f90customize CompaqVisualFCompilerCould not locate executable DFcustomize IntelItaniumVisualFCompilerCould not locate executable eflcustomize Gnu95FCompilerCould not locate executable gfortranCould not locate executable f95customize G95FCompilerCould not locate executable g95customize IntelEM64VisualFCompilercustomize IntelEM64TFCompilerCould not locate executable efortCould not locate executable efccustomize PGroupFlangCompilerCould not locate executable flangdon't know how to compile Fortran code on platform 'nt'NOT AVAILABLEatlas_3_10_blas_threads_info:Setting PTATLAS=ATLASlibraries tatlas not found in ['c:\\octoprint\\venv\\lib', 'C:\\']NOT AVAILABLEatlas_3_10_blas_info:libraries satlas not found in ['c:\\octoprint\\venv\\lib', 'C:\\']NOT AVAILABLEatlas_blas_threads_info:Setting PTATLAS=ATLASlibraries ptf77blas,ptcblas,atlas not found in ['c:\\octoprint\\venv\\lib', 'C:\\']NOT AVAILABLEatlas_blas_info:libraries f77blas,cblas,atlas not found in ['c:\\octoprint\\venv\\lib', 'C:\\']NOT AVAILABLEaccelerate_info:NOT AVAILABLEC:\Users\owner\AppData\Local\Temp\pip-install-7y3db7s3\numpy\numpy\distutils\system_info.py:1914: UserWarning:Optimized (vendor) Blas libraries are not found.Falls back to netlib Blas library which has worse performance.A better performance should be easily gained by switchingBlas library.if self._calc_info(blas):blas_info:libraries blas not found in ['c:\\octoprint\\venv\\lib', 'C:\\']NOT AVAILABLEC:\Users\owner\AppData\Local\Temp\pip-install-7y3db7s3\numpy\numpy\distutils\system_info.py:1914: UserWarning:Blas (http://www.netlib.org/blas/) libraries not found.Directories to search for the libraries can be specified in thenumpy/distutils/site.cfg file (section [blas]) or by settingthe BLAS environment variable.if self._calc_info(blas):blas_src_info:NOT AVAILABLEC:\Users\owner\AppData\Local\Temp\pip-install-7y3db7s3\numpy\numpy\distutils\system_info.py:1914: UserWarning:Blas (http://www.netlib.org/blas/) sources not found.Directories to search for the sources can be specified in thenumpy/distutils/site.cfg file (section [blas_src]) or by settingthe BLAS_SRC environment variable.if self._calc_info(blas):NOT AVAILABLEnon-existing path in 'numpy\\distutils': 'site.cfg'lapack_opt_info:lapack_mkl_info:libraries mkl_rt not found in ['c:\\octoprint\\venv\\lib', 'C:\\']NOT AVAILABLEopenblas_lapack_info:libraries openblas not found in ['c:\\octoprint\\venv\\lib', 'C:\\']NOT AVAILABLEopenblas_clapack_info:libraries openblas,lapack not found in ['c:\\octoprint\\venv\\lib', 'C:\\']NOT AVAILABLEflame_info:libraries flame not found in ['c:\\octoprint\\venv\\lib', 'C:\\']NOT AVAILABLEatlas_3_10_threads_info:Setting PTATLAS=ATLASlibraries lapack_atlas not found in c:\octoprint\venv\liblibraries tatlas,tatlas not found in c:\octoprint\venv\liblibraries lapack_atlas not found in C:\libraries tatlas,tatlas not found in C:\<class 'numpy.distutils.system_info.atlas_3_10_threads_info'>NOT AVAILABLEatlas_3_10_info:libraries lapack_atlas not found in c:\octoprint\venv\liblibraries satlas,satlas not found in c:\octoprint\venv\liblibraries lapack_atlas not found in C:\libraries satlas,satlas not found in C:\<class 'numpy.distutils.system_info.atlas_3_10_info'>NOT AVAILABLEatlas_threads_info:Setting PTATLAS=ATLASlibraries lapack_atlas not found in c:\octoprint\venv\liblibraries ptf77blas,ptcblas,atlas not found in c:\octoprint\venv\liblibraries lapack_atlas not found in C:\libraries ptf77blas,ptcblas,atlas not found in C:\<class 'numpy.distutils.system_info.atlas_threads_info'>NOT AVAILABLEatlas_info:libraries lapack_atlas not found in c:\octoprint\venv\liblibraries f77blas,cblas,atlas not found in c:\octoprint\venv\liblibraries lapack_atlas not found in C:\libraries f77blas,cblas,atlas not found in C:\<class 'numpy.distutils.syste [... cut ...] lstudio.microsoft.com/downloads/----------------------------------------ERROR: Command errored out with exit status 1: 'c:\octoprint\venv\scripts\python.exe' 'c:\octoprint\venv\lib\site-packages\pip\_vendor\pep517\_in_process.py' prepare_metadata_for_build_wheel 'C:\Users\owner\AppData\Local\Temp\tmpejlhstnm' Check the logs for full command output.Preparing wheel metadata: finished with status 'error'Error!Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

steveiebbb, 2021-08-08:
Thank you for the efforts, and I apologize for not doing it sooner. Took Octo out of the equation and can see no mesh values returned after a g29. Issue CAN'T be on the Octo-side. Sorry again, and you rock!

closed 2021-08-08 by steveiebbb