jneilliii/OctoPrint-BedLevelVisualizer #456: [BUG]: Settings do not save after server reboot

DigiVore-Programing, 2021-08-11:
**Describe the bug**
When you change corrections settings, they do not persist after a server / system reboot

**Expected behavior**
All settings staying after any number of reboots or system reboots. 

**Debug Log**
File empty...

**Screenshots**
[image]

**Firmware and Version**
Marlin v2.x.x bugfix (unsure date of bugfix, planning on moving to release Marlin soon.)

jneilliii, 2021-08-11:
could you please enable debut logging in the plugin's settings, restart octoprint, and test again. Then the plugin_bedlevelvisualizer_debug.log file won't be empty. However, I think this might be a front-end UI issue. Could you open your browser's developer tools (F12) and on the console tab look for any red text/errors on initial page load, or when opening the settings.

DigiVore-Programing, 2021-08-12:
Nope. Still empty.

Hayden, Hayden Fixes <http://haydenfixes.jordanmccollum.com>
Repairing iPhones and computers since 2017

DigiVore-Programing, 2021-08-12:
However, it did not clear my settings this time.
Hayden, Hayden Fixes <http://haydenfixes.jordanmccollum.com>
Repairing iPhones and computers since 2017

closed 2021-09-02 by github-actions[bot]