jneilliii/OctoPrint-BedLevelVisualizer #467: [BUG]: Not able to return to printer main menu

sybaris, 2021-09-08:
**Describe the bug**
At the end of an update Mesh, I cannot use again my printer.
I need to shutdown, and restart to be able to use me menu

**Expected behavior**
Be able to return again on my main menu of my ender 3 v2

**Debug Log**

**Screenshots**
At the end of the update Mesh, here my screen :
[image]
When I cancel, I have this popup :
[image]
But when I click on Confirm, nothing happen. If I choose cancel, I come back to the previous screen...

**Firmware and Version**
Ender 3 v2
Jyers 1.3.5

jneilliii, 2021-09-08:
jyers firmware recommends telling the screen that it's done printing. If you add an M77 to the end of your update scripts command does it work?

https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings#gcode-scripts

sybaris, 2021-09-08:
Here my script :

M140 S60 ; starting by heating the bed for nominal mesh accuracy
M117 Homing all axes ; send message to printer display
G28      ; home all axes
M117 Heating the bed ; send message to printer display
M190 S60 ; waiting until the bed is fully warmed up
M300 S1000 P500 ; chirp to indicate bed mesh levels is initializing
M117 Creating the bed mesh levels ; send message to printer display
M155 S30 ; reduce temperature reporting rate to reduce output pollution
@BEDLEVELVISUALIZER	; tell the plugin to watch for reported mesh
G29 T	   ; run bilinear probing
M155 S3  ; reset temperature reporting
M140 S0 ; cooling down the bed
M300 S440 P200 ; make calibration completed tones
M300 S660 P250
M300 S880 P300
M117 Bed mesh levels completed ; send message to printer display

jneilliii, 2021-09-08:
Try this.

```
M75
M140 S60 ; starting by heating the bed for nominal mesh accuracy
M117 Homing all axes ; send message to printer display
G28 ; home all axes
M117 Heating the bed ; send message to printer display
M190 S60 ; waiting until the bed is fully warmed up
M300 S1000 P500 ; chirp to indicate bed mesh levels is initializing
M117 Creating the bed mesh levels ; send message to printer display
M155 S30 ; reduce temperature reporting rate to reduce output pollution
@BEDLEVELVISUALIZER ; tell the plugin to watch for reported mesh
G29 T ; run bilinear probing
M155 S3 ; reset temperature reporting
M140 S0 ; cooling down the bed
M300 S440 P200 ; make calibration completed tones
M300 S660 P250
M300 S880 P300
M117 Bed mesh levels completed ; send message to printer display
M77
```

sybaris, 2021-09-09:
Great, now it works :-)
Problem solved.
Thanks

closed 2021-09-09 by sybaris