jneilliii/OctoPrint-BedLevelVisualizer #468: [FR]: Changeable colors of CSS classes .raise and .lower

electronicm, 2021-09-10:
**Is your feature request related to a problem? Please describe.**
I use a dark theme and the blue of the CSS class `.raise` is very hard to read, as well as the red of the class `.lower`.

**Describe the solution you'd like**
editable otions for the both classes as colorpicker or textfield for hex/colorname.

**Describe alternatives you've considered**
Right now I'm using some custom css to override your color. `.meshtable .raise { color: #006aff !important; }`

**Additional context**

jneilliii, 2021-09-10:
Your approach of using the custom/advanced css configurations of whatever UI customization plugin you are using is spot on, and probably the best approach for now, as it's only because of the other plugin that this looks non-optimal.

electronicm, 2021-09-10:
BTW: I am using UI Customizer with the theme Discorded.

jneilliii, 2021-09-10:
Yeah, same would work with any theme or themeify. I'm leaning towards just documenting the class names and showing examples of your workaround because the ui is already getting cluttered.

PR #497 (2022-01-01): 1.1.1
* add ability to configure graph height in visualization settings, resolves #462
* add min, max, and variance values of graphed mesh, resolves #286
* add OctoDash support, create Custom Action with `[!WEB]http://127.0.0.1:5000/plugin/bedlevelvisualizer/bedlevelvisualizer` as the command.
* adjust the tables for better theme support, resolves #479
* add option for showing mesh data on tab, resolves #496
* remove tooltip hover on tables, resolves #490
* add Prusa adjustment values to graph (still needs verification), resolves #242
* add appearance > title setting to automatically downloade [... cut]
closed 2022-01-01 by jneilliii