jneilliii/OctoPrint-BedLevelVisualizer #469: [FR]: lets do some math for raise/lower

electronicm, 2021-09-10:
**Is your feature request related to a problem? Please describe.**
It's just a "nice to have" and would make my life a little easier.
But it could increase the usability of the plugin for leveling the bed enormously.

**Describe the solution you'd like**
Most with ABL/UBL use different scales on their level screws.Here it would be good if the output value in the correction matrix, or in the hover, corresponded to the respective scale.

For example, the standard level screws of the Ender series have a graduation that corresponds to approx. 30° per graduation mark.
If the user could now enter a formula like 'X/30', he would see directly by how many graduation lines he has to adjust the wheel.

**Describe alternatives you've considered**

1. I choose one of the four relevant values as the zero point, 
2. take the degree number from the correction matrix for the other three points, 
3. divide each by 30 and 
4. round it down.

Only when the value is greater than one does an adjustment really make sense.

**Additional context**
BTW: I have dimensioned my UBL mash so that 4 points are almost above the level screws.

jneilliii, 2021-09-10:
I'm sorry, but isn't the adjustment angle options exactly for that use case and tells you how much of a turn to make?

electronicm, 2021-09-10:
As far as yes, but my thought goes one step further.
When I have the number of degrees, I have to convert it to my scale every time.
My idea was to let the plugin do this conversion right away. So that I can immediately see by how many graduation marks I have to turn the screw.

Example: I have a value of 114 °. I then have to divide this by 30 °/division to get the value of 3.8 divisions and then round it off.

If it were possible to store the number of partial reports or degrees per partial area in the plugin, it would show 3 or 3.8 directly.
This would improve the user experience through intuitive usability.

My value of 30 ° per sub-area could still be calculated in my head to some extent. Others may have 20 sub-ranges, which corresponds to 18°.  There it becomes a little more difficult with mental arithmetic.

Do you understand what I mean?

jneilliii, 2021-09-10:
I think I understand but am not familiar with the scale your referring to. Maybe a photo? Seems very specific to your printer.

electronicm, 2021-09-11:
[Take this wheel, for example.](https://www.thingiverse.com/thing:4393673)
It has 16 peaks which could be used as a 'scale'.
This would then be 360 ​​° / 16 peaks = 22.5 ° per division/peak.

If you had to turn the wheel 114°, the plugin could simply display a value of 5 peaks or graduation marks.

Sorry that my English is not so great.

jneilliii, 2021-09-11:
I understand now, thanks. Will consider for a future update. I've been wanting to redo how the adjustment table is done anyway, but because I didn't add that functionality of the plugin and it's very complicated how it works it may be a while before I get around to it.

shoorek83, 2022-02-26:
I second that. That would be the most useful feature and would save a lot of time spent converting correction data.

closed 2025-06-22 by electronicm