jneilliii/OctoPrint-BedLevelVisualizer #470: [FR]: Auto calculate bed variance

SirChryse, 2021-09-12:
**Is your feature request related to a problem? Please describe.**
No it is not

**Describe the solution you'd like**
It would be nice to have something on the graph show the calculated bed variance, even though I learned how to calculate it there are some users who may not know how to calculate it and this would be convenient for them. 

**Describe alternatives you've considered**
n/a

**Additional context**
n/a

jneilliii, 2021-09-18:
Also include min/max values somewhere.

jneilliii, 2021-09-18:
Implemented in above commit and available in version 1.1.1rc2. Change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted if you want to test it out. 

[image]

closed 2022-01-01 by jneilliii