jneilliii/OctoPrint-BedLevelVisualizer #471: [BUG]: data not correct after moving bed in z-axis

dpeart, 2021-09-15:
I'm using this on a Snapmaker 2.0 printer

The plugin worked fine until I changed how my bed attaches to the printer carriage.  I added springs between the bed and the carriage to provide a better leveling mechanism.  This resulted in the bed being raised by ~18mm, reducing the overall build height by that amount.

What I get now is the first probe point is ~10mm higher than expected, making for a bad result.

[image]

I have rebooted everything, I did a clean bed leveling using the Snapmaker feature and still get this issue.  I can visually see that the bed is not tilted by 10mm!

Attached is the log file.
[plugin_bedlevelvisualizer_debug (4).log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/7171180/plugin_bedlevelvisualizer_debug.4.log)

Suggestion?

jneilliii, 2021-09-15:
The plugin merely graphs what it reported by the firmware. If you use the relative z option it could help, but only assuming that one outlier point gets corrected.

dpeart, 2021-09-15:
Would changing the build volume in the plugin make any difference?

jneilliii, 2021-09-15:
It won't make a difference. I don't know why your firmware is reporting such large numbers though. My understanding from discussions in the OctoPrint Discord server is that usually happens if your z offset isn't set prior to probing. Just take a look at the numbers reported by firmware....very strange.

```
[2021-09-15 15:52:06,760] DEBUG: ['+38.05875', '+34.27472', '+30.49069', '+27.79250', '+26.90403', '+27.10139', '+27.29875']
[2021-09-15 15:52:06,766] DEBUG: ['+34.24176', '+31.86516', '+29.48856', '+27.81250', '+27.30402', '+27.49608', '+27.68815']
[2021-09-15 15:52:06,774] DEBUG: ['+30.42476', '+29.45559', '+28.48641', '+27.83250', '+27.70401', '+27.89077', '+28.07754']
[2021-09-15 15:52:06,780] DEBUG: ['+27.75125', '+27.75764', '+27.76402', '+27.81500', '+27.94028', '+28.11013', '+28.28000']
[2021-09-15 15:52:06,785] DEBUG: ['+26.98352', '+27.24570', '+27.50788', '+27.73500', '+27.90368', '+28.03728', '+28.17088']
[2021-09-15 15:52:06,794] DEBUG: ['+27.35926', '+27.44537', '+27.53148', '+27.61750', '+27.70334', '+27.78907', '+27.87481']
[2021-09-15 15:52:06,799] DEBUG: ['+27.73500', '+27.64505', '+27.55509', '+27.50000', '+27.50301', '+27.54088', '+27.57875']
[2021-09-15 15:52:06,806] DEBUG: {'type': 'rectangular', 'x_min': 0, 'x_max': 320.0, 'y_min': 0, 'y_max': 350.0, 'z_min': 0, 'z_max': 330.0}
[2021-09-15 15:52:06,806] DEBUG: stopping mesh collection
```

jneilliii, 2021-09-15:
Also, are you homing with an endstop or using the probe for homing?

dpeart, 2021-09-15:
Homing is with end stops.

jneilliii, 2021-09-15:
Sorry, I should be more specific. Are you homing z with an endstop? That seems to be a potential reason that the numbers would be so large is because the z probe offset doesn't apply to homing.

dpeart, 2021-09-17:
I believe it was an issue with the sensor providing bad readings on the initial probe.  Got it all sorted.

thanks

closed 2021-09-17 by dpeart