jneilliii/OctoPrint-BedLevelVisualizer #472: [BUG]: no downgrade at release channel change

electronicm, 2021-09-18:
**Describe the bug**
When I change the release channel from "Release Candidate" back to "Stable", I do not get a downgrade.
I have to uninstall and reinstall the plug in.

**Expected behavior**
When I change the release channel, as with various other plugins and also OctoPrint itself, the current version of the channel should be installed.

jneilliii, 2021-09-18:
This should work like the others, Are there any errors in octoprint.log or plugin_bedlevelvisualizer_debug.log file?

jneilliii, 2021-09-19:
confirmed reproducible

jneilliii, 2021-09-19:
I believe this to be an upstream bug and have created the above issue in OctoPrint's repo to track.

jneilliii, 2022-01-01:
Since this appears to be resolved upstream and will be release in the next version of OctoPrint closing issue.

closed 2022-01-01 by jneilliii