jneilliii/OctoPrint-BedLevelVisualizer #473: [BUG]: first column of the X-axis is missing

electronicm, 2021-09-18:
**Describe the bug**
I have entered 7 mesh points for the X-axis and these are also output on the `G29 T` or `M420 V`.
However, only columns 1-6 are displayed by the plugin under "Data" and "Corrections".

**Expected behavior**
I would also like to see column 0 of the X axis. :D

**Debug Log**
plugin_bedlevelvisualizer_debug.log
```
[2021-09-18 21:35:46,087] DEBUG: mesh collection started
[2021-09-18 21:35:46,092] DEBUG: mesh collection started
[2021-09-18 21:35:46,129] DEBUG: []
[2021-09-18 21:35:46,227] DEBUG: ['+0.070', '+0.015', '-0.040', '-0.058', '-0.048', '-0.055', '-0.055']
[2021-09-18 21:35:46,230] DEBUG: []
[2021-09-18 21:35:46,278] DEBUG: ['+0.070', '+0.015', '-0.040', '-0.058', '-0.048', '-0.055', '-0.055']
[2021-09-18 21:35:46,281] DEBUG: []
[2021-09-18 21:35:46,329] DEBUG: ['+0.090', '+0.055', '+0.020', '+0.002', '-0.005', '-0.020', '-0.020']
[2021-09-18 21:35:46,332] DEBUG: []
[2021-09-18 21:35:46,391] DEBUG: ['+0.062', '+0.042', '+0.022', '-0.002', '+0.010', '-0.002', '-0.002']
[2021-09-18 21:35:46,394] DEBUG: []
[2021-09-18 21:35:46,444] DEBUG: ['+0.065', '+0.045', '+0.025', '-0.005', '+0.010', '-0.005', '-0.005']
[2021-09-18 21:35:46,447] DEBUG: []
[2021-09-18 21:35:46,496] DEBUG: ['+0.074', '+0.072', '+0.071', '+0.032', '+0.050', '+0.032', '+0.032']
[2021-09-18 21:35:46,499] DEBUG: []
[2021-09-18 21:35:46,554] DEBUG: ['+0.112', '+0.099', '+0.085', '+0.082', '+0.080', '+0.070', '+0.070']
[2021-09-18 21:35:46,557] DEBUG: []
[2021-09-18 21:35:46,607] DEBUG: ['+0.060', '+0.040', '+0.020', '+0.027', '+0.005', '-0.015', '-0.015']
[2021-09-18 21:35:46,610] DEBUG: []
[2021-09-18 21:35:46,666] DEBUG: ['+0.060', '+0.040', '+0.020', '+0.027', '+0.005', '-0.015', '-0.015']
[2021-09-18 21:35:46,677] DEBUG: {'type': 'rectangular', 'x_min': 0.0, 'x_max': 231.0, 'y_min': 0.0, 'y_max': 221.0, 'z_min': 0.0, 'z_max': 250.0}
[2021-09-18 21:35:46,677] DEBUG: stopping mesh collection
[2021-09-18 21:35:46,679] DEBUG: flipped y axis:
[2021-09-18 21:35:46,679] DEBUG: ['+0.015', '-0.040', '-0.058', '-0.048', '-0.055', '-0.055']
[2021-09-18 21:35:46,680] DEBUG: ['+0.015', '-0.040', '-0.058', '-0.048', '-0.055', '-0.055']
[2021-09-18 21:35:46,680] DEBUG: ['+0.055', '+0.020', '+0.002', '-0.005', '-0.020', '-0.020']
[2021-09-18 21:35:46,680] DEBUG: ['+0.042', '+0.022', '-0.002', '+0.010', '-0.002', '-0.002']
[2021-09-18 21:35:46,681] DEBUG: ['+0.045', '+0.025', '-0.005', '+0.010', '-0.005', '-0.005']
[2021-09-18 21:35:46,681] DEBUG: ['+0.072', '+0.071', '+0.032', '+0.050', '+0.032', '+0.032']
[2021-09-18 21:35:46,681] DEBUG: ['+0.099', '+0.085', '+0.082', '+0.080', '+0.070', '+0.070']
[2021-09-18 21:35:46,682] DEBUG: ['+0.040', '+0.020', '+0.027', '+0.005', '-0.015', '-0.015']
[2021-09-18 21:35:46,682] DEBUG: ['+0.040', '+0.020', '+0.027', '+0.005', '-0.015', '-0.015']
[2021-09-18 21:35:46,683] DEBUG: Final mesh:
[2021-09-18 21:35:46,684] DEBUG: ['+0.015', '-0.040', '-0.058', '-0.048', '-0.055', '-0.055']
[2021-09-18 21:35:46,684] DEBUG: ['+0.015', '-0.040', '-0.058', '-0.048', '-0.055', '-0.055']
[2021-09-18 21:35:46,684] DEBUG: ['+0.055', '+0.020', '+0.002', '-0.005', '-0.020', '-0.020']
[2021-09-18 21:35:46,685] DEBUG: ['+0.042', '+0.022', '-0.002', '+0.010', '-0.002', '-0.002']
[2021-09-18 21:35:46,685] DEBUG: ['+0.045', '+0.025', '-0.005', '+0.010', '-0.005', '-0.005']
[2021-09-18 21:35:46,685] DEBUG: ['+0.072', '+0.071', '+0.032', '+0.050', '+0.032', '+0.032']
[2021-09-18 21:35:46,686] DEBUG: ['+0.099', '+0.085', '+0.082', '+0.080', '+0.070', '+0.070']
[2021-09-18 21:35:46,686] DEBUG: ['+0.040', '+0.020', '+0.027', '+0.005', '-0.015', '-0.015']
[2021-09-18 21:35:46,686] DEBUG: ['+0.040', '+0.020', '+0.027', '+0.005', '-0.015', '-0.015']
```

octoprint.log
```
2021-09-18 21:35:46,133 - octoprint.util.comm - ERROR - Error while processing hook bedlevelvisualizer:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 3983, in _readline
    ret = hook(self, ret)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 274, in process_gcode
    new_line.pop(0)
IndexError: pop from empty list
2021-09-18 21:35:46,231 - octoprint.util.comm - ERROR - Error while processing hook bedlevelvisualizer:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 3983, in _readline
    ret = hook(self, ret)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
[... 51 lines cut ...]
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 3983, in _readline
    ret = hook(self, ret)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 274, in process_gcode
    new_line.pop(0)
IndexError: pop from empty list
2021-09-18 21:35:46,612 - octoprint.util.comm - ERROR - Error while processing hook bedlevelvisualizer:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 3983, in _readline
    ret = hook(self, ret)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 274, in process_gcode
    new_line.pop(0)
IndexError: pop from empty list
```

**Screenshots**
[image]
[image]

**Firmware and Version**
Marlin bugfix-2.0.x (Sep 18 2021 19:29:44)
It can also be reproduced with the current stable.

jneilliii, 2021-09-18:
Your firmware does a 7x9 grid? I think this may be part of the problem. I remember at one point there was an issue with unequal grid dimensions causing issues.

electronicm, 2021-09-19:
I configured the 7x9 grid to get as close as possible to the bed level screws.
I test it again with 9x9 / 7x7 and give feedback.

Can I do anything else to help?

electronicm, 2021-09-19:
So, I have now made a test with a 7x7 grid. The result is the same.

jneilliii, 2021-09-19:
I need you to enable serial logging in octoprint's settings, restart octoprint, and update mesh once, then disable serial logging after. Then provide ther serial.log.

electronicm, 2021-09-19:
I probably won't make it until next weekend, but you'll get the log.

electronicm, 2021-09-25:
here it is: [serial.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/7230009/serial.log)

electronicm, 2021-09-25:
btw: here is also the debug log
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/7230029/plugin_bedlevelvisualizer_debug.log)

jneilliii, 2021-09-25:
Do you have `remove row labels` enabled on the collection tab in the plugin's settings?

electronicm, 2021-09-25:
Yes

My configuration:
```
  bedlevelvisualizer:
    _config_version: 1
    camera_position: 0,-2,0.3
    colorscale: '[[0, "rebeccapurple"],[0.4, "rebeccapurple"],[0.45, "#006aff"],[0.5,
      "green"],[0.55, "yellow"],[0.6, "red"],[1, "red"]]'
    command: "@BEDLEVELVISUALIZER\t; tell the plugin to watch for reported mesh\n\
      M420 V\t\t\t\t\t; Mesh ausgeben"
    commands:
    - command: "@BEDLEVELVISUALIZER\t; tell the plugin to watch for reported mesh\n\
        G28 O\nG29 J2\t\t\t\t\t; Checking the mesh using n² points\n@BEDLEVELVISUALIZER\t\
        ; tell the plugin to watch for reported mesh\nM420 V\t\t\t\t\t; Mesh ausgeben"
      confirmation: false
      enabled_while_graphing: true
      enabled_while_printing: false
      icon: fas fa-vector-square
      input: []
      label: ''
      message: ''
      tooltip: 4 Point tilt probing
    - command: "@BEDLEVELVISUALIZER\t; tell the plugin to watch for reported mesh\n\
        G28 O\nG29 P1\nG29 P3\nG29 P3\n@BEDLEVELVISUALIZER\t; tell the plugin to watch\
        \ for reported mesh\nG29 S%(SLOT)s T0 V4 \t\t; Mesh speichern und ausgeben\n\
        M500"
      confirmation: true
      enabled_while_graphing: true
      enabled_while_printing: false
      icon: fas fa-file
      input:
      - label: hideme
        parameter: SLOT
        value: '0'
      label: ''
      message: Dadurch wird das Mesh neu erfasst und der gewählten Slot überschrieben.
      tooltip: Das Mesh wird komplett neu erstellt und in den gewählten Slot gespeichert
    - command: "@BEDLEVELVISUALIZER\t; tell the plugin to watch for reported mesh\n\
        G29 L%(SLOT)s\n@BEDLEVELVISUALIZER\t; tell the plugin to watch for reported\
        \ mesh\nM420 V\t\t\t\t\t; Mesh ausgeben"
      confirmation: false
      enabled_while_graphing: true
      enabled_while_printing: false
      icon: fas fa-file-import
      input:
      - label: hideme
        parameter: SLOT
        value: '0'
      label: ''
      message: ''
      tooltip: Das Mesh aus Slot wird geladen
    - command: 'G29 S%(SLOT)s\nM500'
      confirmation: true
      enabled_while_graphing: true
      enabled_while_printing: false
      icon: fas fa-file-export
      input:
      - label: hideme
        parameter: SLOT
        value: '0'
      label: ''
      message: Der gewählte Slot wird überschrieben.
      tooltip: Speichert das aktuelle Mesh im ausgewählten Slot.
    descending_y: true
    ignore_correction_matrix: true
    mesh_timestamp: 25.9.2021, 18:39:23
    rotation: '0'
    screw_hub: 0.7
    showdegree: true
    stored_mesh:
    [REMOVED]
    stored_mesh_x:
    [REMOVED]
    stored_mesh_y:
    [REMOVED]
    stored_mesh_z_height: 250
    stripFirst: true
```

jneilliii, 2021-09-25:
What happens if you disable that option and update mesh again?

electronicm, 2021-09-26:
Yes, you are obviously on the right track.
If I disable the option and update the mesh, I get all the values for the X axis. When I enable it, the first column is missing again. This can be reproduced over and over again.

Tested it for 7x7 and 9x7 grids with the same result.

You don't need to fix the bug immediately because of me.
I now know and have figured out a workaround.

In case you want to document it for others:
For now, I've hidden the line via CSS.
```
div#tab_plugin_bedlevelvisualizer div.meshtable tfoot tr:nth-of-type(0n+2) {
   display:none;
   visibility: hidden;
}
```
`0n+2` hides only the second element, preventing future possible `tr` in `tfoot` from being accidentally hidden.

jneilliii, 2021-09-26:
> Yes, you are obviously on the right track.
> If I disable the option and update the mesh, I get all the values for the X axis. When I enable it, the first column is missing again. This can be reproduced over and over again.

That is the point, if you have that option enabled it's suppose to remove the first column, otherwise leave it unchecked, so I'm not exactly sure what you are trying to accomplish with the css to hide things?