jneilliii/OctoPrint-BedLevelVisualizer #474: Mesh shows about 7mm tilt and warp - dial gauge shows about 1mm:

coupepete, 2021-09-28:
**Describe the bug**
The mesh generated is showing excessive tilt and warping of my bed. Total warp is about 7mm displayed in visualiser, with about 3mm warp i the worst case. 

**Expected behavior**
Maximum measured is about 0.6 mm tilt and 1mm warp, but anyway is it physically much lower. I measured this by mounting a dial gauge to to the extruder and moving it across various points and axis. It seems that the firmware is compensatin correctly for the out of level by testing with a first layer print. ->

**Debug Log**
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/7242993/plugin_bedlevelvisualizer_debug.log)

*[image]

**Firmware and Version**
Octoprint ver 1.6.1  Bed Visualiser 1.1.0 Marlin 2.0.7.2

Thanks really appreciate your work - hopefully this is not a biggy.

jneilliii, 2021-09-28:
The plugin only graphs what your firmware reports for offsets of printed points. Look at the returned data from your printer when doing the process on the terminal tab and you'll see what I mean.

coupepete, 2021-10-13:
Thanks the issue is fixed now. I needed to restore my data from eeprom after updating Marlin to use the bl touch. Sorry for the inconvenience