jneilliii/OctoPrint-BedLevelVisualizer #475: Pinter timeout and disconnecting while mesh probing in progress

alexw22, 2021-10-05:
I am experiencing the same issue as described [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/370) in issue #370. My printer disconnects from octopi while doing the bed probing. I have a 5x5 mesh as its quite a large build plate I have.

I have verified that my G29 command is in the Long Running Command section of Firmware & protocol > Advanced Options as described there and my entries look just like the screenshot in #370.

What parameter setting in octoprint would be adjusting the timeout so that I can increase it. I have tried my luck however without success until now. (I have been increasing: Serial connection -> Intervals & timeouts -> Communication Timeout -> busy protocol support not detected to 300s however no luck until now) I am able to overcome the issue manually by toggling my part cooling fan on and off while probing in order for the probing finish and mesh to be generated properly.

I would highly appreciate any feedback on this :)

jneilliii, 2021-10-05:
Jyers firmware?

alexw22, 2021-10-05:
Marlin 2.0.7.2
I hope this is what you were asking for.

jneilliii, 2021-10-05:
Yeah, there was a known issue with jyers fork of marlin with leveling that would cause problems like you describe. If you haven't installed that then probably not the issue. Do you know what type of auto bed leveling is enabled? Basically it's your printer that stops responding during the process, causing octoprint to think the printer is dead. Other possible causes is your firmware expects an sd card in the printer to save eeprom data. Only way to truly diagnose is enable serial logging in octoprint's settings and try the process again, then share your serial.log file. That will tell us for sure if it's firmware or not.

alexw22, 2021-10-05:
To my knowledge I did not install jyers. I just installed Marlin of the official website and edited the config to fit my needs/printer. I am using a genuine BlTouch sensor.

I enabled serial logging and re-ran the bed probing procedure. It disconnected while the bed was still heating up. I don't think however that it has anything to do with the heating/power. I'm expecting it to be just some timer that disconnects octoprint after a set amount of time. Im sure if i reran it now it would manage to heat up to 60 and start probing and disconnect while probing (what I was describing previously).

Hopefully this gives you enough information. I am happy to provide any other information if necessary.

`2021-10-05 17:52:16,371 - Enabling serial logging
2021-10-05 17:52:17,353 - Recv: Not SD printing
2021-10-05 17:52:18,308 - Recv:  T:19.64 /0.00 B:23.24 /0.00 @:0 B@:0
2021-10-05 17:52:18,353 - Recv: Not SD printing
2021-10-05 17:52:19,353 - Recv: Not SD printing
2021-10-05 17:52:20,308 - Recv:  T:20.20 /0.00 B:22.77 /0.00 @:0 B@:0
2021-10-05 17:52:20,353 - Recv: Not SD printing
2021-10-05 17:52:21,353 - Recv: Not SD printing
2021-10-05 17:52:22,307 - Recv:  T:19.87 /0.00 B:22.85 /0.00 @:0 B@:0
2021-10-05 17:52:22,353 - Recv: Not SD printing
2021-10-05 17:52:23,353 - Recv: Not SD printing
2021-10-05 17:52:24,307 - Recv:  T:19.53 /0.00 B:22.89 /0.00 @:0 B@:0
2021-10-05 17:52:24,353 - Recv: Not SD printing
2021-10-05 17:52:25,353 - Recv: Not SD printing
2021-10-05 17:52:26,307 - Recv:  T:19.87 /0.00 B:22.89 /0.00 @:0 B@:0
2021-10-05 17:52:26,353 - Recv: Not SD printing
2021-10-05 17:52:27,353 - Recv: Not SD printing
2021-10-05 17:52:28,307 - Recv:  T:19.53 /0.00 B:22.85 /0.00 @:0 B@:0
2021-10-05 17:52:28,353 - Recv: Not SD printing
2021-10-05 17:52:29,353 - Recv: Not SD printing
2021-10-05 17:52:30,307 - Recv:  T:19.75 /0.00 B:22.81 /0.00 @:0 B@:0
2021-10-05 17:52:30,353 - Recv: Not SD printing
2021-10-05 17:52:31,353 - Recv: Not SD printing
2021-10-05 17:52:32,308 - Recv:  T:19.98 /0.00 B:22.97 /0.00 @:0 B@:0
2021-10-05 17:52:32,352 - Recv: Not SD printing
2021-10-05 17:52:33,353 - Recv: Not SD printing
2021-10-05 17:52:34,308 - Recv:  T:20.31 /0.00 B:23.01 /0.00 @:0 B@:0
2021-10-05 17:52:34,366 - Recv: Not SD printing
2021-10-05 17:52:35,366 - Recv: Not SD printing
2021-10-05 17:52:36,307 - Recv:  T:20.31 /0.00 B:22.93 /0.00 @:0 B@:0
2021-10-05 17:52:36,365 - Recv: Not SD printing
2021-10-05 17:52:36,666 - Send: M140 S60
2021-10-05 17:52:36,681 - Recv: ok P15 B3
2021-10-05 17:52:36,689 - Send: M117 Homing all axes
2021-10-05 17:52:36,705 - Recv: //action:notification Homing all axes
2021-10-05 17:52:36,724 - Recv: ok P15 B3
2021-10-05 17:52:36,729 - Send: G28
2021-10-05 17:52:37,366 - Recv: Not SD printing
2021-10-05 17:52:38,312 - Recv:  T:19.53 /0.00 B:22.81 /60.00 @:0 B@:127
2021-10-05 17:52:38,365 - Recv: Not SD printing
2021-10-05 17:52:38,735 - Recv: echo:busy: processing
2021-10-05 17:52:38,737 - Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
2021-10-05 17:52:39,366 - Recv: Not SD printing
2021-10-05 17:52:40,307 - Recv:  T:19.98 /0.00 B:22.85 /60.00 @:0 B@:127
2021-10-05 17:52:40,366 - Recv: Not SD printing
2021-10-05 17:52:40,734 - Recv: echo:busy: processing
2021-10-05 17:52:41,366 - Recv: Not SD printing
2021-10-05 17:52:42,307 - Recv:  T:19.64 /0.00 B:23.32 /60.00 @:0 B@:127
2021-10-05 17:52:42,366 - Recv: Not SD printing
2021-10-05 17:52:42,734 - Recv: echo:busy: processing
2021-10-05 17:52:43,365 - Recv: Not SD printing
2021-10-05 17:52:44,307 - Recv:  T:19.31 /0.00 B:23.67 /60.00 @:0 B@:127
2021-10-05 17:52:44,366 - Recv: Not SD printing
2021-10-05 17:52:45,747 - Recv: echo:busy: processing
2021-10-05 17:52:45,757 - Recv: Not SD printing
2021-10-05 17:52:46,307 - Recv:  T:19.31 /0.00 B:24.30 /60.00 @:0 B@:127
2021-10-05 17:52:46,750 - Recv: Not SD printing
2021-10-05 17:52:47,748 - Recv: echo:busy: processing
2021-10-05 17:52:47,756 - Recv: Not SD printing
2021-10-05 17:52:48,308 - Recv:  T:19.75 /0.00 B:25.09 /60.00 @:0 B@:127
2021-10-05 17:52:48,749 - Recv: Not SD printing
2021-10-05 17:52:51,102 - Recv: echo:busy: processing
2021-10-05 17:52:51,112 - Recv:  T:19.64 /0.00 B:25.60 /60.00 @:0 B@:127
2021-10-05 17:52:51,115 - Recv: Not SD printing
[... 318 lines cut ...]
2021-10-05 17:54:41,761 - Recv:  T:19.08 /0.00 B:46.88 /60.00 @:0 B@:127
2021-10-05 17:54:41,804 - Recv: Not SD printing
2021-10-05 17:54:41,808 - Recv: echo:busy: processing
2021-10-05 17:54:41,899 - Recv:  T:19.87 /0.00 B:46.92 /60.00 @:0 B@:127 W:?
2021-10-05 17:54:42,806 - Recv: Not SD printing
2021-10-05 17:54:42,900 - Recv:  T:19.31 /0.00 B:46.97 /60.00 @:0 B@:127 W:?
2021-10-05 17:54:43,761 - Recv:  T:19.08 /0.00 B:47.27 /60.00 @:0 B@:127
2021-10-05 17:54:43,804 - Recv: Not SD printing
2021-10-05 17:54:43,808 - Recv: echo:busy: processing
2021-10-05 17:54:43,899 - Recv:  T:19.53 /0.00 B:47.15 /60.00 @:0 B@:127 W:?
2021-10-05 17:54:44,806 - Recv: Not SD printing
2021-10-05 17:54:44,900 - Recv:  T:19.53 /0.00 B:47.30 /60.00 @:0 B@:127 W:?
2021-10-05 17:54:45,760 - Recv:  T:19.75 /0.00 B:47.42 /60.00 @:0 B@:127
2021-10-05 17:54:45,805 - Recv: Not SD printing
2021-10-05 17:54:45,808 - Recv: echo:busy: processing
2021-10-05 17:54:45,900 - Recv:  T:19.53 /0.00 B:47.57 /60.00 @:0 B@:127 W:?
2021-10-05 17:54:46,805 - Recv: Not SD printing
2021-10-05 17:54:46,899 - Recv:  T:19.53 /0.00 B:47.73 /60.00 @:0 B@:127 W:?
2021-10-05 17:54:47,760 - Recv:  T:19.75 /0.00 B:47.76 /60.00 @:0 B@:127
2021-10-05 17:54:47,805 - Recv: Not SD printing
2021-10-05 17:54:47,807 - Recv: echo:busy: processing
2021-10-05 17:54:47,900 - Recv:  T:19.53 /0.00 B:47.76 /60.00 @:0 B@:127 W:?
2021-10-05 17:54:48,805 - Recv: Not SD printing
2021-10-05 17:54:48,831 - Connection closed, closing down monitor
2021-10-05 17:54:48,835 - Changing monitoring state from "Operational" to "Offline"`

alexw22, 2021-10-05:
Have done that just dat. It however did not make a difference unfortunately.

jneilliii, 2021-10-05:
And it disconnects during the heat up process? Does it do the same if you send those commands from the terminal tab? Try with just G29 instead of G29 T maybe.

alexw22, 2021-10-05:
If the bed is cold yes. And it does not advance to probing the bed. I imagine because the disconnect happens before the bed probing GCODE is sent. Then If i reconnect and cancel and restart the bed probing process it warms up and starts probing and disconnects while probing. Finishes the probing process but does not display anything because while octoprint is already disconnected it never receives the bed probing data.

I will give G29 instead of G29 T a try now and edit this post when i have the results

alexw22, 2021-10-05:
I did receive this error message just now after finishing another calibration print:

Bed Visualizer Error
Timeout occured before processing completed. Processing may still be running or there may be a configuration error. Consider increasing the Processing Timeout value in settings and restart OctoPrint.

Wondering what Processing Timeout value in settings it is referring to!

alexw22, 2021-10-06:
When I preheat before a print, it never disconnects though. I don't understand how this is different.

alexw22, 2021-10-06:
While running G29 manually in the octoprint terminal tab, it does not disconnect.

jneilliii, 2021-10-06:
And what does it return at the end of the process?

alexw22, 2021-10-06:
It did return the bed matrix in the terminal after it finished. If you need the serial output I would have to re-run the command manually tomorrow. Please let me know if you would like me to do that.

So when running G29 manually it finishes without disconnects and appears to return the expected output in the terminal tab. When I run the GCODE sequence (I posed above) through the plugin it disconnects after some amount of time while heating the bed/probing.

jneilliii, 2021-10-06:
That's what is confusing me. The way the plugin works is basically doing the same thing as sending the commands manually. Let's simplify your settings. Replace the collection commands with just the following. 

```
G28
@BEDLEVELVISUALIZER
G29
```

Manually heat the bed to desired temp, and then attempt the mesh collection. Let's see where we get with that. Then we can slowly add in one command at a time to that to see what command is causing the issue.

alexw22, 2021-10-06:
I changed the list of commands to the 3 commands above.
The printer still disconnected while heating up the bed manually! But I tired one more time and reconnected and continued to heat up the bed to 60 after which I started the probing with the 3 commands from above. It again disconnected while probing.
However what is interesting is that when starting a print file within octoprint through the web interface (which obviously heats up the bed before starting to print) it does **not** cause a disconnect while heating up the bed/printing. I am still very puzzled by what is going on.

jneilliii, 2021-10-06:
My best guess is that some timeout is happening in firmware. I'm not 100% sure why, but this is obviously not a plugin problem if OctoPrint is disconnecting while heating up manually. Sorry I couldn't be more help, I suggest hitting up the forum or OctoPrint discord about the disconnect during heating.

closed 2021-10-06 by jneilliii

alexw22, 2021-10-06:
TLDR: Check your other plugins related to power management. In my case it was the "PSU Control" plugin that had its variables changed after a plugin update that changed some variables.

I managed to narrow it down to what the issue here is. I am just including it here in case anyone is experiencing something similar. Exactly What you are saying. This lead me to believe it must be in the firmware because it seems like it is happening outside of the plugin. I checked all of the firmware line by line to look for any timeouts. I didn't find any but then it the idea came up that it could be another plugin that can cause the issue. So I checked the plugins that were installed earlier.
I recently updated octoprint and all plugins. For some reason the "PSU control" plugin reset some values and caused the printer to disconnect after 2 minutes. Because it appears that more variables were changed after the update the PSU control did not manage to turn the printer off but did disconnect.

Thank you jneilliii for your support and efforts in troubleshooting this issue!